platformio/platform-raspberrypi #1: Large program fails; shows strange LED blink pattern

jhmaloney, 2021-04-24:
MicroBlocks is a virtual machine (like MicroPython) that runs on a wide range of 32-bit microcontrollers. We use PlatformIO to build for all of them. It is a fairly large program.

I am porting MicroBlocks to the Raspberry Pi RP2040 chip, the Pico board in particular. It compiles and runs using Arduino IDE 1.8.13 (Mac OS) and Arduino's "Mbed OS RP2040" 2.0.0 board package. The same code compiles and install it with PlatformIO, but the program does not run on the board. The board does not appear as serial port (which it does when the program runs correctly), and the LED blinks in a strange pattern: four long blinks followed by three quick ones. That looks like it might indicating an error code of some sort but I haven't been able to find any documentation that will help me decode it.

Here's the platformio.ini file entry I'm using:

[env:pico]
platform = https://github.com/platformio/platform-raspberrypi.git
framework = arduino
board = pico
lib_ignore = WiFi
build_flags = -D ARDUINO_RASPBERRY_PI_PICO

jhmaloney, 2021-04-24:
Correction: MicroBlocks does NOT run correctly when built with Arduino's "Mbed OS RP2040" 2.0.0 board package. It does run when built with [this](https://github.com/earlephilhower/arduino-pico) community supported package.

So this is Arduino's bug, not a PlatformIO bug.

I'm still curious about the strange LED blink pattern, though. Any idea what it means or where it is documented?

jhmaloney, 2021-04-24:
That blink pattern seems to indicate an assertion failure in the mbed initialization sequence. The blink patter is the same as the one shown in the animated GIF in [this thread](https://community.platformio.org/t/stm32f103c8-mbed-support/12302/7).

It's not clear what assertion is failing, however.

jhmaloney, 2021-04-26:
Update: The issue was a misaligned memory access. Our code includes an array of uint8_t that is *also* accessed by word. Other compilers were word-aligning the start of that array, but not this compiler. Adding "__attribute__ ((aligned (32)))" solved the problem.

I'm fairly sure that the funny blink pattern was coming from mbed's exception handler, but I'm still not sure if the blink pattern encodes the error type. I'd love to know how to get more information about such mbed errors in case I run into any more of them in the future.

valeros, 2021-04-26:
Hi @jhmaloney ! Thanks for keeping us posted. 

> Update: The issue was a misaligned memory access. Our code includes an array of uint8_t that is also accessed by word. Other compilers were word-aligning the start of that array, but not this compiler. Adding "attribute ((aligned (32)))" solved the problem.

Could you please explain what you meant by `other compilers`? Other versions or something completely different like Clang?

> I'm fairly sure that the funny blink pattern was coming from mbed's exception handler, but I'm still not sure if the blink pattern encodes the error type. I'd love to know how to get more information about such mbed errors in case I run into any more of them in the future.

I guess the pattern you saw is  so-called `Lights of Death`, more info here https://os.mbed.com/handbook/Debugging#runtime-errors

jhmaloney, 2021-04-27:
Thanks! I'd found the "Lights of Death" page but the blink pattern on the Pico is different what is shown the right-most board on that page (with a single LED). I suspect mbed changed the pattern but hasn't updated the page. The animated gif [here](https://community.platformio.org/t/stm32f103c8-mbed-support/12302/7) shows that new blink pattern.

By other compilers, I mean the compilers used in the all other Arduino PlatformIO frameworks that we use to build MicroBlocks including compilers for nRF5x, SAMD21, ESP8266, and ESP32, Teensy, and more. We've been using PlatformIO to build MicroBlocks for several dozen platforms and this is the first compiler that hasn't aligned the start of that uint8_t array on a word boundary. Admittedly, since I hadn't included a "align 32" pragma, the compiler is free to do that.

So, this isn't a compiler bug, it is just different behavior than previous compilers. (The data structure is accessed as both 32-bit words and as unsigned bytes. It was declared as a uint8_t array because that was convenient and resulted in fewer typecasts.)

In case others run into this when porting working code code from another microcontroller to to the RP2040, the morale of the story is if you get the "lights of death" you might want to double-check your data structure alignments.

Incidentally, the memory map file showed all my global variables as being at address 0. That probably *is* a compiler bug.