platformio/platform-raspberrypi #2: Upload port required

danielkucera, 2021-04-26:
If I don't specify upload port, I get an error message:
```
Processing pico (platform: raspberrypi; board: pico; framework: arduino)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (0.0.0+sha.aa251bf) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, raspberrypi-swd, stlink)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - tool-openocd-raspberrypi 2.1100.0 (11.0) 
 - tool-rp2040tools 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 47 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 50500 bytes from 270336 bytes)
Flash: [          ]   3.5% (used 73056 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, raspberrypi-swd, stlink
Looking for upload port...

Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/page/faq.html#platformio-udev-rules

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
===================================================================== [FAILED] Took 1.12 seconds ====
```

If it's really the case, it would be good to mention this in the example or docs.

MNS26, 2021-04-26:
have you found a solution?
what did you specify?
im also having the same problem.

danielkucera, 2021-04-27:
I'm running on Ubuntu and if I insert the board in upload mode, it presents itself as a USB drive so I added:
```
upload_port = /media/danman/RPI-RP2
```
which is the standard path to mounted drives in Ubuntu

MNS26, 2021-04-27:
ah ok i just tried it in a linux vm and yeah that worked `/media/MS25/RPI-RP2`
although I now have it also working on windows but in both cases it fails at the last part (the dotted line)

[image]
probably a different issue?

valeros, 2021-04-27:
Hi @MNS26 ! Does a COM port show up in Device Manager when you connect your board board?

MNS26, 2021-04-27:
yes it does show up 
[image]

valeros, 2021-04-27:
@MNS26  Could you please check VID:PID of the COM10 port?

MNS26, 2021-04-27:
the VID is 2e8a and the PID is 00c0
[image]

valeros, 2021-04-27:
@MNS26 That's strange, the VID:PID combination in the board manifest is the same. Could you please run the `pio run -t upload -v` in the IDE terminal and copy paste the output here?

MNS26, 2021-04-27:
@valeros here is the output 

```
PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter> pio run -t upload -v
Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; board: pico; framework: arduino; upload_port: R:/; lib_deps: )
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (0.0.0+sha.aa251bf) (git+https://github.com/platformio/platform-raspberrypi.git) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, raspberrypi-swd, stlink)
PACKAGES:
 - framework-arduino-mbed 2.0.0
 - tool-openocd-raspberrypi 2.1100.0 (11.0)
 - tool-rp2040tools 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\LittleVGL
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\Portenta_SDCARD
Found 27 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
No dependencies
Building in release mode
MethodWrapper(["checkprogsize"], [".pio\build\pico\firmware.elf"])
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 50500 bytes from 270336 bytes)
Flash: [          ]   3.5% (used 72640 bytes from 2097152 bytes)
.pio\build\pico\firmware.elf  :

section                               size        addr

.boot2                                 256   268435456

.text                                52640   268435712

.rodata                               8620   268488352

.ARM.exidx                               8   268496972

.ram_vector_table                      192   536870912

.data                                11372   536871104

.uninitialized_ram.magic_location       12   536882476

.uninitialized_data                      0   536882488

.scratch_x                               0   537133056

.scratch_y                               0   537137152

.bss                                 39128   536882488

.heap                               210416   536921616

.ARM.attributes                         40           0

.comment                               132           0

.debug_frame                          5144           0

.stab                                   60           0

.stabstr                               118           0

Total                               328138
<lambda>(["upload"], [".pio\build\pico\firmware.elf"])
AVAILABLE: cmsis-dap, raspberrypi-swd, stlink
BeforeUpload(["upload"], [".pio\build\pico\firmware.elf"])
Use manually specified: R:/
Forcing reset using 1200bps open/close on port R:/
C:\Users\MS26\.platformio\packages\tool-rp2040tools\rp2040load -v -D .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
================================================================================================ [FAILED] Took 13.34 seconds ================================================================================================
PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter> 
```

MNS26, 2021-04-27:
i dont see any obvious errors 🤔

valeros, 2021-04-27:
Forgot to mention, please delete the custom upload port as well.

MNS26, 2021-04-27:
@valeros 
this time it also reset the pico but did not exit bootsel(?) mode
[image]

removed the manual port 
```
PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter> pio run -t upload -v
Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; board: pico; framework: arduino; lib_deps: )
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (0.0.0+sha.aa251bf) (git+https://github.com/platformio/platform-raspberrypi.git) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, raspberrypi-swd, stlink)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - tool-openocd-raspberrypi 2.1100.0 (11.0) 
 - tool-rp2040tools 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\LittleVGL
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\Portenta_SDCARD
Found 27 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio\build\pico\src\main.cpp.o -c -Wvla -fno-rtti -std=gnu++14 -DMBED_TRAP_ERRORS_ENABLED=1 -Os -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m0plus -mthumb -iprefixC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino @C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\includes.txt -nostdlib -DPLATFORMIO=50101 -DRASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DARM_MATH_CM0PLUS -D__CMSIS_RTOS -DCOMPONENT_FLASHIAP=1 -D__CORTEX_M0PLUS -DDEVICE_ANALOGIN=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_PORT_IN=1 -DDEVICE_PORT_OUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -D__MBED__=1 -DMBED_BUILD_TIMESTAMP=1617899306.9586625 -D__MBED_CMSIS_RTOS_CM -DMBED_MPU_CUSTOM -DPICO_NO_BINARY_INFO=1 -DPICO_ON_DEVICE=1 -DPICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1 -DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED -DPICO_UART_ENABLE_CRLF_SUPPORT=0 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M0 -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_memmap_default -DTARGET_NAME=RASPBERRY_PI_PICO -DTARGET_RASPBERRYPI -DTARGET_RASPBERRY_PI_PICO -DTARGET_RELEASE -DTARGET_RP2040 -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_NO_GLOBAL_USING_DIRECTIVE=1 -DUSE_ARDUINO_PINOUT -DARDUINO=10810 -DARDUINO_ARCH_MBED -Iinclude -Isrc -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino\api\deprecated -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO src\main.cpp
arm-none-eabi-gcc -o .pio\build\pico\FrameworkArduinoVariant\double_tap_usb_boot.c.o -c -std=gnu11 -DMBED_TRAP_ERRORS_ENABLED=1 -Os -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m0plus -mthumb -iprefixC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino @C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\includes.txt -nostdlib -DPLATFORMIO=50101 -DRASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DARM_MATH_CM0PLUS -D__CMSIS_RTOS -DCOMPONENT_FLASHIAP=1 -D__CORTEX_M0PLUS -DDEVICE_ANALOGIN=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_PORT_IN=1 -DDEVICE_PORT_OUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -D__MBED__=1 -DMBED_BUILD_TIMESTAMP=1617899306.9586625 -D__MBED_CMSIS_RTOS_CM -DMBED_MPU_CUSTOM -DPICO_NO_BINARY_INFO=1 -DPICO_ON_DEVICE=1 -DPICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1 -DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED -DPICO_UART_ENABLE_CRLF_SUPPORT=0 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M0 -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_memmap_default -DTARGET_NAME=RASPBERRY_PI_PICO -DTARGET_RASPBERRYPI -DTARGET_RASPBERRY_PI_PICO -DTARGET_RELEASE -DTARGET_RP2040 -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_NO_GLOBAL_USING_DIRECTIVE=1 -DUSE_ARDUINO_PINOUT -DARDUINO=10810 -DARDUINO_ARCH_MBED -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino\api\deprecated -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\double_tap_usb_boot.c
[... cut ...]
Forcing reset using 1200bps open/close on port COM10
C:\Users\MS26\.platformio\packages\tool-rp2040tools\rp2040load -v -D .pio\build\pico\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
================================================================================================ [FAILED] Took 22.57 seconds ================================================================================================
PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter>
```

valeros, 2021-04-27:
@MNS26 According to your screenshot above, the COM10 port is the right one, but it fails for some reason. Could you please also try to upload a simple blink example using Arduino IDE? How the upload workflow implemented there? Would be great if you could copy paste here the logs from Arduino IDE. Please keep in mind that you might need to enable the verbose log in the Settings:
[image]

MNS26, 2021-04-27:
@valeros i used the stand-alone version of arduino (dont plan on installing it)
it successfully uploaded the default blink example

```
C:\Users\MS26\Desktop\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\MS26\Desktop\arduino-nightly\hardware -hardware C:\Users\MS26\AppData\Local\Arduino15\packages -tools C:\Users\MS26\Desktop\arduino-nightly\tools-builder -tools C:\Users\MS26\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\MS26\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\MS26\Desktop\arduino-nightly\libraries -libraries C:\Users\MS26\Documents\Arduino\libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=2E8A_00C0 -ide-version=10814 -build-path C:\Users\MS26\AppData\Local\Temp\arduino_build_942219 -warnings=all -build-cache C:\Users\MS26\AppData\Local\Temp\arduino_cache_956469 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.rp2040tools.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.openocd.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.dfu-util.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -verbose C:\Users\MS26\Desktop\arduino-nightly\examples\01.Basics\Blink\Blink.ino
C:\Users\MS26\Desktop\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\MS26\Desktop\arduino-nightly\hardware -hardware C:\Users\MS26\AppData\Local\Arduino15\packages -tools C:\Users\MS26\Desktop\arduino-nightly\tools-builder -tools C:\Users\MS26\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\MS26\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\MS26\Desktop\arduino-nightly\libraries -libraries C:\Users\MS26\Documents\Arduino\libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=2E8A_00C0 -ide-version=10814 -build-path C:\Users\MS26\AppData\Local\Temp\arduino_build_942219 -warnings=all -build-cache C:\Users\MS26\AppData\Local\Temp\arduino_cache_956469 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.rp2040tools.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.openocd.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.dfu-util.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -verbose C:\Users\MS26\Desktop\arduino-nightly\examples\01.Basics\Blink\Blink.ino
Using board 'pico' from platform in folder: C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.0.0
Using core 'arduino' from platform in folder: C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.0.0
Detecting libraries used...
"C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO/defines.txt" "@C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO/cxxflags.txt" -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10814 -DARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_MBED_RP2040 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\cores\\arduino" "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO" "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15 [... cut ...]
Sketch uses 80072 bytes (0%) of program storage space. Maximum is 16777216 bytes.
Global variables use 52920 bytes (19%) of dynamic memory, leaving 217416 bytes for local variables. Maximum is 270336 bytes.
Forcing reset using 1200bps open/close on port COM10
C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2/rp2040load -v -D C:\Users\MS26\AppData\Local\Temp\arduino_build_942219/Blink.ino.elf 
rp2040load 1.0.1 - compiled with go1.15.8
Loading into Flash: [==============================]  100%
```

valeros, 2021-04-27:
Does it work if you manually specify COM10 as the upload port, e.g.:
```
upload_port = COM10
```

MNS26, 2021-04-27:
@valeros just ran it with 
`upload_port = COM10`

and now it worked
```
PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter> pio run -t upload -v
Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; board: pico; framework: arduino; upload_port: COM10; lib_deps: )
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (0.0.0+sha.aa251bf) (git+https://github.com/platformio/platform-raspberrypi.git) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, raspberrypi-swd, stlink)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - tool-openocd-raspberrypi 2.1100.0 (11.0) 
 - tool-rp2040tools 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\LittleVGL
Framework incompatible library C:\Users\MS26\.platformio\packages\framework-arduino-mbed\libraries\Portenta_SDCARD
Found 27 compatible libraries
More details about "Library Compatibility Mode": https://docs.platformio.org/page/librarymanager/ldf.html#ldf-compat-mode
Scanning dependencies...
No dependencies
Building in release mode
arm-none-eabi-g++ -o .pio\build\pico\src\main.cpp.o -c -Wvla -fno-rtti -std=gnu++14 -DMBED_TRAP_ERRORS_ENABLED=1 -Os -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m0plus -mthumb -iprefixC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino @C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\includes.txt -nostdlib -DPLATFORMIO=50101 -DRASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DARM_MATH_CM0PLUS -D__CMSIS_RTOS -DCOMPONENT_FLASHIAP=1 -D__CORTEX_M0PLUS -DDEVICE_ANALOGIN=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_PORT_IN=1 -DDEVICE_PORT_OUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -D__MBED__=1 -DMBED_BUILD_TIMESTAMP=1617899306.9586625 -D__MBED_CMSIS_RTOS_CM -DMBED_MPU_CUSTOM -DPICO_NO_BINARY_INFO=1 -DPICO_ON_DEVICE=1 -DPICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1 -DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED -DPICO_UART_ENABLE_CRLF_SUPPORT=0 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M0 -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_memmap_default -DTARGET_NAME=RASPBERRY_PI_PICO -DTARGET_RASPBERRYPI -DTARGET_RASPBERRY_PI_PICO -DTARGET_RELEASE -DTARGET_RP2040 -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_NO_GLOBAL_USING_DIRECTIVE=1 -DUSE_ARDUINO_PINOUT -DARDUINO=10810 -DARDUINO_ARCH_MBED -Iinclude -Isrc -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino\api\deprecated -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO src\main.cpp
arm-none-eabi-gcc -o .pio\build\pico\FrameworkArduinoVariant\double_tap_usb_boot.c.o -c -std=gnu11 -DMBED_TRAP_ERRORS_ENABLED=1 -Os -Wall -Wextra -Wno-missing-field-initializers -Wno-unused-parameter -c -fdata-sections -ffunction-sections -fmessage-length=0 -fno-exceptions -fomit-frame-pointer -funsigned-char -mcpu=cortex-m0plus -mthumb -iprefixC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino @C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\includes.txt -nostdlib -DPLATFORMIO=50101 -DRASPBERRY_PI_PICO -DARDUINO_ARCH_RP2040 -DARM_MATH_CM0PLUS -D__CMSIS_RTOS -DCOMPONENT_FLASHIAP=1 -D__CORTEX_M0PLUS -DDEVICE_ANALOGIN=1 -DDEVICE_FLASH=1 -DDEVICE_I2C=1 -DDEVICE_I2CSLAVE=1 -DDEVICE_INTERRUPTIN=1 -DDEVICE_PORT_IN=1 -DDEVICE_PORT_OUT=1 -DDEVICE_PWMOUT=1 -DDEVICE_RESET_REASON=1 -DDEVICE_SERIAL=1 -DDEVICE_SERIAL_FC=1 -DDEVICE_SPI=1 -DDEVICE_USBDEVICE=1 -DDEVICE_USTICKER=1 -DDEVICE_WATCHDOG=1 -D__MBED__=1 -DMBED_BUILD_TIMESTAMP=1617899306.9586625 -D__MBED_CMSIS_RTOS_CM -DMBED_MPU_CUSTOM -DPICO_NO_BINARY_INFO=1 -DPICO_ON_DEVICE=1 -DPICO_RP2040_USB_DEVICE_ENUMERATION_FIX=1 -DPICO_TIME_DEFAULT_ALARM_POOL_DISABLED -DPICO_UART_ENABLE_CRLF_SUPPORT=0 -DTARGET_CORTEX -DTARGET_CORTEX_M -DTARGET_LIKE_CORTEX_M0 -DTARGET_LIKE_MBED -DTARGET_M0P -DTARGET_memmap_default -DTARGET_NAME=RASPBERRY_PI_PICO -DTARGET_RASPBERRYPI -DTARGET_RASPBERRY_PI_PICO -DTARGET_RELEASE -DTARGET_RP2040 -DTOOLCHAIN_GCC -DTOOLCHAIN_GCC_ARM -DMBED_NO_GLOBAL_USING_DIRECTIVE=1 -DUSE_ARDUINO_PINOUT -DARDUINO=10810 -DARDUINO_ARCH_MBED -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\cores\arduino\api\deprecated -IC:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO C:\Users\MS26\.platformio\packages\framework-arduino-mbed\variants\RASPBERRY_PI_PICO\double_tap_usb_boot.c
[... cut ...]
Loading into Flash: [=========================     ]  84%
Loading into Flash: [==========================    ]  89%
Loading into Flash: [============================  ]  95%
Loading into Flash: [==============================]  100%

================================================================================================ [SUCCESS] Took 12.16 seconds ================================================================================================PS C:\Users\MS26\Documents\PlatformIO\Projects\Raspberry pi pico helicopter> 
```

MNS26, 2021-04-27:
also knowing that the pico has 2 cores is there a way to use the 2nd one? and if so how?

and when i tried to upload something again via vscode it froze
[image]

MNS26, 2021-04-27:
@valeros  _maybe a little off topic for this issue_

but when i was using the unofficial (WizIO) platform
i was able to fade the builtin led with this

```
int brightness;
int fadeAmount =3;

void setup()
{
  Serial.begin(9600);
  pinMode(25, OUTPUT);
}

void loop()
{
  analogWrite(25,map(brightness,0,255,0,65535));
  brightness = brightness + fadeAmount;
  // reverse the direction of the fading at the ends of the fade:
  if (brightness <= 0 || brightness >= 255) {
    fadeAmount = -fadeAmount;
  }
 Serial.println(map(brightness,0,255,0,65535));
delay(30);
}
```

but now with the exact same code it just turns of for a split second roughly every second

MNS26, 2021-04-27:
nvm i just rewrote the exact same code  and now it works -_- . . . _why just why_

savejeff, 2021-04-27:
I got it to work on windows using the upload port setting 

```

[env:pico]
;platform = https://github.com/platformio/platform-raspberrypi.git
platform = raspberrypi
framework = arduino
board = pico
upload_port = D:\
```

MNS26, 2021-04-28:
@danielkucera i think you can close this issue by now seeing that its basically solved

danielkucera, 2021-04-28:
But it does NOT work for me without upload_port (on Ubuntu).

MNS26, 2021-04-28:
nvm then

maxgerhardt, 2021-04-28:
I wasn't able to use @savejeff 's method (it would just timeout with `.......` shown), I had to use Zadig to install libusb-win32 drivers for the "RP2 Boot2 (Interface 1)" device so that the upload would work. Otherwise it would also show me an unknown device in the device manager.

Maybe for people who already used the Arduino IDE with the Pico have this driver pre-installed?

On Windows 10 btw.

savejeff, 2021-04-30:
> Maybe for people who already used the Arduino IDE with the Pico have this driver pre-installed?
FYI : I have installed the Pico Core in Arduino IDE before I installed it on PlatformIO. That might be the reason I have no problems with Driver etc.

mpmarks, 2021-05-09:
I had problems uploading on Ubuntu. It turned out that picotool was having protection problems writing to the usb disk. I solved that by adding the line:

SUBSYSTEMS=="usb", ATTRS{idVendor}=="2e8a", MODE:="0666"

to /etc/udev/rules.d/99-platformio-udev.rules

So my platformio.ini for Ubuntu is:

[env:pico]
platform = raspberrypi
board = pico
framework = arduino
upload_port = /media/<my username>/RPI-RP2
upload_protocol = picotool

Hope this is useful.

cscott, 2021-07-07:
Debian system here.  Also having the mysterious failures.  Using @mpmarks addition to my udev rules helped, even though I'd installed the latest udev rules from  https://docs.platformio.org/en/latest/faq.html#platformio-udev-rules and *even though I could upload successfully by manually copying to /media/USERNAME/RPI-RP2*.  Ie, it wasn't obvious this was a permissions error, but it was.  

Now I have no upload-specific information in my `platformio.ini` (ie, using the default `upload_protocol = picotool` for `platform = raspberrypi` and `board = pico`, and no explicit specification of an `upload_port`), and it *mostly* works.

An arduino sketch is already running on my pico, so the autoreset works.

But it's still not perfect: the first time I run `pio -t upload` I get:
```
[...]
Checking size .pio/build/snitch/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.9% (used 51076 bytes from 270336 bytes)
Flash: [          ]   3.6% (used 75424 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Uploading .pio/build/snitch/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
========================= [FAILED] Took 11.83 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
snitch         FAILED    00:00:11.832
==================== 1 failed, 0 succeeded in 00:00:11.832 ====================
```
But I can tell that the forced reset works and my sketch has stopped running.  Re-running `pio run -t upload` the second time I get:
```
[...]
Checking size .pio/build/snitch/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.9% (used 51076 bytes from 270336 bytes)
Flash: [          ]   3.6% (used 75424 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Uploading .pio/build/snitch/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
...Loading into Flash: [                              ]  0%
Loading into Flash: [=                             ]  5%
Loading into Flash: [===                           ]  10%
Loading into Flash: [====                          ]  16%
Loading into Flash: [======                        ]  21%
Loading into Flash: [========                      ]  27%
Loading into Flash: [=========                     ]  32%
Loading into Flash: [===========                   ]  37%
Loading into Flash: [============                  ]  43%
Loading into Flash: [==============                ]  48%
Loading into Flash: [================              ]  54%
Loading into Flash: [=================             ]  59%
Loading into Flash: [===================           ]  64%
Loading into Flash: [=====================         ]  70%
Loading into Flash: [======================        ]  75%
Loading into Flash: [========================      ]  81%
Loading into Flash: [=========================     ]  86%
Loading into Flash: [===========================   ]  91%
Loading into Flash: [============================= ]  97%
Loading into Flash: [==============================]  100%

========================= [SUCCESS] Took 3.97 seconds =========================

Environment    Status    Duration
-------------  --------  ------------
snitch         SUCCESS   00:00:03.970
========================= 1 succeeded in 00:00:03.970 =========================
```

So this seems like the remaining issue is just a race condition: it's resetting correctly but not waiting *quite* long enough to let the upload succeed.  The second time I run the upload commands it is already reset into BOOTSEL mode and (now that the `udev` file has been updated) the upload succeeds.

Is there any way to tweak the wait time between reset and upload?

csivanich, 2021-09-14:
> ...
> SUBSYSTEMS=="usb", ATTRS{idVendor}=="2e8a", MODE:="0666"
> 
> to /etc/udev/rules.d/99-platformio-udev.rules
> 
> So my platformio.ini for Ubuntu is:
> 
> [env:pico]
> platform = raspberrypi
> board = pico
> framework = arduino
> upload_port = /media//RPI-RP2
> upload_protocol = picotool
> 
> Hope this is useful.

After adding the above mentioned udev rule (and a reboot), my build uploads are now working without issue after seeing the timeout `......` on VS Code + PIO. Thanks @mpmarks!

My platform.io, for reference. I was able to just use the serial port for upload, and this removes need for mounting too.

    [env:pico]
    platform = raspberrypi
    board = pico
    framework = arduino
    upload_protocol = picotool
    upload_port = /dev/ttyACM0

And an upload run:

```
> Executing task in folder Pi Pico Testing: platformio run --target upload <

Processing pico (platform: raspberrypi; board: pico; framework: arduino)
---------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.3.0) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed 2.4.1 
 - tool-openocd-raspberrypi 2.1100.0 (11.0) 
 - tool-rp2040tools 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 30 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.6% (used 50240 bytes from 270336 bytes)
Flash: [          ]   3.5% (used 72744 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Use manually specified: /dev/ttyACM0
Forcing reset using 1200bps open/close on port /dev/ttyACM0
Uploading .pio/build/pico/firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
...Loading into Flash: [                              ]  0%
Loading into Flash: [=                             ]  5%
Loading into Flash: [===                           ]  11%
Loading into Flash: [====                          ]  16%
Loading into Flash: [======                        ]  22%
Loading into Flash: [========                      ]  27%
Loading into Flash: [=========                     ]  33%
Loading into Flash: [===========                   ]  39%
Loading into Flash: [=============                 ]  44%
Loading into Flash: [===============               ]  50%
Loading into Flash: [================              ]  55%
Loading into Flash: [==================            ]  61%
Loading into Flash: [====================          ]  67%
Loading into Flash: [=====================         ]  72%
Loading into Flash: [=======================       ]  78%
Loading into Flash: [========================      ]  83%
Loading into Flash: [==========================    ]  89%
Loading into Flash: [============================  ]  95%
Loading into Flash: [==============================]  100%

============== [SUCCESS] Took 3.08 seconds ==============

Terminal will be reused by tasks, press any key to close it.
```

Running on Arch Linux/5.13.13 as an unprivileged user in `plugdev` group

jcw, 2021-11-03:
Pico-noob here, pio 5.2.3b4, macos 12.0.1.

FWIW, the *only* way I got things to work smoothly was as follows:

* plug Pico in with boot button pressed
* include `upload_port = /Volumes/RPI-RP2`
* no `upload_protocol` defined, it defaults to `picotool`
* create a blink sketch, include `Serial.begin(9600);` in `setup()`
* `pio run -t upload`
* the on-board LED blinks
* *comment out* the `upload_protocol` line
* now `pio run -t upload` will work without boot/reset button hassles

The `System Information` app can be used to see what mode the Pico is in:
* in the USB Device Tree tab, "RP2 Boot" or "RaspberryPi Pico" is reported

It's not a great out-of-the-box first-time experience, but at least it works.
I suspect that a crashing app will require going through the above setup again.

chechu2k5, 2021-12-18:
I'm no `go` expert but I went over the `rp2040tools` [source code](https://github.com/arduino/rp2040tools/blob/master/main.go) and noticed it had little to no error handling.
I've [modified](https://github.com/chechu2k5/rp2040tools) it to include some extra verbose logging and error handling (still not ideal but should be better). 
I built the `rp2040tools.exe` using `go build` and replaced the one in `.platformio\packages\tool-rp2040tools` after backing up the original. I am able to upload the program from within VSCode like so (note the extra logging info at the end). 

> I only tested this on Windows 10, but I don't think would be issues with my changes on other platforms.

```
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 31 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Checking size .pio\build\pico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  15.1% (used 40696 bytes from 270336 bytes)
Flash: [          ]   0.2% (used 4034 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: COM19
Forcing reset using 1200bps open/close on port COM19
Uploading .pio\build\pico\firmware.elf
rp2040load 0.0.0-dev - compiled with go1.17.5
Converting elf to uf2 ... OK
Looking for RP2040 device in BOOTSEL mode . OK
Loading into Flash: [                              ]  0%
Loading into Flash: [=                             ]  5%
Loading into Flash: [===                           ]  11%
Loading into Flash: [====                          ]  16%
Loading into Flash: [======                        ]  22%
Loading into Flash: [========                      ]  28%
Loading into Flash: [=========                     ]  33%
Loading into Flash: [===========                   ]  39%
Loading into Flash: [=============                 ]  44%
Loading into Flash: [===============               ]  50%
Loading into Flash: [================              ]  56%
Loading into Flash: [==================            ]  61%
Loading into Flash: [====================          ]  67%
Loading into Flash: [=====================         ]  72%
Loading into Flash: [=======================       ]  78%
Loading into Flash: [=========================     ]  84%
Loading into Flash: [==========================    ]  89%
Loading into Flash: [============================  ]  95%
Loading into Flash: [==============================]  100%

Rebooting RP2040 device ... OK
```

Hopefully, anyone encountering the `...` failure will be able see where the failure happens with my changes.

kylarosborne, 2022-04-03:
I tried the above Go fix, but it did not work for me.
I did solve the problem by editing rp2040load Go program. The change I made was: 
Remove all commands after the "Converting elf to uf2" (elf2uf2 command)
Use io.Copy to copy the firmware.elf.uf2 to D:\ (which is where the raspberry pi pico opens for my computer)

This was my first time using Go. Hope this helps someone with Pico upload in the future.

asukiaaa, 2022-05-15:
On my environment, I need to run the copy command for uploading firmware after showing `/dev/${USER}/}/RPI-RP2/`
```
cp .pio/build/pico/firmware.elf.uf2 /media/${USER}/RPI-RP2/
```

If `/dev/${USER}/}/RPI-RP2/` is not shown, upload command shows it and failed so I need to run command like this.
```
pio run -t upload && cp .pio/build/pico/firmware.elf.uf2 /media/${USER}/RPI-RP2/
```

Should the picotool upload uf2 file to the media?

My environment
Ubuntu: 20.04
PlatformIO: 5.2.5

maxgerhardt, 2022-05-15:
>Should the picotool upload uf2 file to the media?

No it does so via libusb.

I have a patch to get `upload_protocol = mbed` working properly, meaning it will copy the .uf2 file (and not the .bin file) to the USB drive, see https://github.com/maxgerhardt/platform-raspberrypi/commit/27cbf996200cb79842d845a899ba8c52127ad269 CC @valeros.

asukiaaa, 2022-05-15:
Thank you for the information but I cannot upload with the following settings.

```
[env:pico]
platform = git@github.com:maxgerhardt/platform-raspberrypi.git#develop
board = pico
framework = arduino
upload_protocol = mbed
```

```
Building in release mode
AttributeError: 'SConsEnvironment' object has no attribute 'UploadUF2ToDisk':
  File "/home/asuki/.local/lib/python3.8/site-packages/platformio/builder/main.py", line 181:
    env.SConscript("$BUILD_SCRIPT")
  File "/home/asuki/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 597:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/home/asuki/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Script/SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/home/asuki/.platformio/platforms/raspberrypi@src-6078dcc2be76b29adbdcbcc3e8dec6e3/builder/main.py", line 300:
    env.VerboseAction(env.UploadUF2ToDisk, "Uploading $SOURCE")
================================ [FAILED] Took 3.16 seconds ================================
```

maxgerhardt, 2022-05-15:
Oops, can you do a `pio platform update "git@github.com:maxgerhardt/platform-raspberrypi.git#develop"` in a [CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and retry? (Or delete all  `/home/asuki/.platformio/platforms/raspberrypi*` folders). I've pushed a hopefully working fix.

asukiaaa, 2022-05-15:
Thank you for updating but it still failed for both case for /devtty/ACM0 available and /media/${USER}/RPI-RP2 available.

```
Looking for upload disk...
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
================================ [FAILED] Took 1.10 seconds ================================
```

maxgerhardt, 2022-05-15:
Hmm auto-detection of that can still be.. imrpoved then. Setting `upload_port =  /media/${USER}/RPI-RP2` (with user replaced) makes it work?

asukiaaa, 2022-05-15:
Thank you for the advise but `upload_port = /media/${USER}/RPI-RP2` cause error for both cases (/dev/ttyACM0 and /media/${USER)/RPI-RP2.

```
Traceback (most recent call last):
  File "/home/asuki/.platformio/packages/tool-scons/scons-local-4.3.0/SCons/Action.py", line 1279, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
TypeError: UploadUF2ToDisk() missing 1 required positional argument: '_'
```

maxgerhardt, 2022-05-15:
Try updating the platform per above again and retry, sorry.

asukiaaa, 2022-05-16:
The log become succeeded but it does not upload firmware.
Can you create a pull request?
I want to report on it.
```
Use manually specified: /media/asuki/RPI-RP2
Uploading .pio/build/pico/firmware.bin
Firmware file /home/asuki/pio/test-gauge/.pio/build/pico/firmware..uf2 not found.

=============================== [SUCCESS] Took 2.39 seconds ===============================
```

maxgerhardt, 2022-05-16:
Sorry, please try updating again and retry. If it still not works, issue to https://github.com/maxgerhardt/platform-raspberrypi/issues

fiechr, 2022-05-29:
> Sorry, please try updating again and retry. If it still not works, issue to https://github.com/maxgerhardt/platform-raspberrypi/issues

Since I'm having the same issue, I added a reply in the existing issue:
https://github.com/maxgerhardt/platform-raspberrypi/issues/5#issuecomment-1140312633

FritschBalazs, 2024-11-17:
If someone is still having this issue on win 11 this is what finally worked for me: [toonrobotics blog post](https://toonrobotics.com/en/win11_platio_2022_10_20en/)

TLDR: you need to update the driver with zadig, as mentioned by @maxgerhardt above, but you have to install winusb (so not libusb_win32) for `RP2 boot` device.