platformio/platform-raspberrypi #3: unable to use 2nd or more serial ports

MNS26, 2021-04-28:
when trying to use a GPS module (bn-880 to be precise) there is no data on the 2nd serial (is it properly implemented?)

the the picture i connected the gps module AFTER i got the check connection message
[image]

the code i used
```
#include <Arduino.h>
#include <TinyGPS++.h>
#include "I2Cdev.h"
#include "MPU6050.h"
#include <PID_v1.h>
//#include "header.h"
#include "math.h"

static const uint32_t GPSBaud = 38400;

// The TinyGPS++ object
TinyGPSPlus gps;

void setup()
{
  Serial.begin(115200);
  Serial1.begin(GPSBaud);

  Serial.println(F("Sats HDOP  Latitude   Longitude   Fix  Date       Time     Date Alt    Course Speed Card  Distance Course Card  Chars Sentences Checksum"));
  Serial.println(F("           (deg)      (deg)       Age                      Age  (m)    --- from GPS ----  ---- to London  ----  RX    RX        Fail"));
  Serial.println(F("----------------------------------------------------------------------------------------------------------------------------------------"));
}

// This custom version of delay() ensures that the gps object
// is being "fed".
static void smartDelay(unsigned long ms)
{
  unsigned long start = millis();
  do 
  {
    while (Serial1.available())
      gps.encode(Serial1.read());
  } while (millis() - start < ms);
}

static void printFloat(float val, bool valid, int len, int prec)
{
  if (!valid)
  {
    while (len-- > 1)
      Serial.print('*');
    Serial.print(' ');
  }
  else
  {
    Serial.print(val, prec);
    int vi = abs((int)val);
    int flen = prec + (val < 0.0 ? 2 : 1); // . and -
    flen += vi >= 1000 ? 4 : vi >= 100 ? 3 : vi >= 10 ? 2 : 1;
    for (int i=flen; i<len; ++i)
      Serial.print(' ');
  }
  smartDelay(0);
}

static void printInt(unsigned long val, bool valid, int len)
{
  char sz[32] = "*****************";
  if (valid)
    sprintf(sz, "%ld", val);
  sz[len] = 0;
  for (int i=strlen(sz); i<len; ++i)
    sz[i] = ' ';
  if (len > 0) 
    sz[len-1] = ' ';
  Serial.print(sz);
  smartDelay(0);
}

static void printDateTime(TinyGPSDate &d, TinyGPSTime &t)
{
  if (!d.isValid())
  {
    Serial.print(F("********** "));
  }
  else
  {
    char sz[32];
    sprintf(sz, "%02d/%02d/%02d ", d.month(), d.day(), d.year());
    Serial.print(sz);
  }
  
  if (!t.isValid())
  {
    Serial.print(F("******** "));
  }
  else
  {
    char sz[32];
    sprintf(sz, "%02d:%02d:%02d ", t.hour(), t.minute(), t.second());
    Serial.print(sz);
  }

  printInt(d.age(), d.isValid(), 5);
  smartDelay(0);
}

static void printStr(const char *str, int len)
{
  int slen = strlen(str);
  for (int i=0; i<len; ++i)
    Serial.print(i<slen ? str[i] : ' ');
  smartDelay(0);
}

void loop()
{
  static const double AMSTERDAM_LAT = 52.370216, AMSTERDAM_LON = 4.895168;

  printInt(gps.satellites.value(), gps.satellites.isValid(), 5);
  printFloat(gps.hdop.hdop(), gps.hdop.isValid(), 6, 1);
  printFloat(gps.location.lat(), gps.location.isValid(), 11, 6);
  printFloat(gps.location.lng(), gps.location.isValid(), 12, 6);
  printInt(gps.location.age(), gps.location.isValid(), 5);
  printDateTime(gps.date, gps.time);
  printFloat(gps.altitude.meters(), gps.altitude.isValid(), 7, 2);
  printFloat(gps.course.deg(), gps.course.isValid(), 7, 2);
  printFloat(gps.speed.kmph(), gps.speed.isValid(), 6, 2);
  printStr(gps.course.isValid() ? TinyGPSPlus::cardinal(gps.course.deg()) : "*** ", 6);

  unsigned long distanceKmToLondon =
    (unsigned long)TinyGPSPlus::distanceBetween(
      gps.location.lat(),
      gps.location.lng(),
      AMSTERDAM_LAT, 
      AMSTERDAM_LON) / 1000;
  printInt(distanceKmToLondon, gps.location.isValid(), 9);

  double courseToLondon =
    TinyGPSPlus::courseTo(
      gps.location.lat(),
      gps.location.lng(),
      AMSTERDAM_LAT, 
      AMSTERDAM_LON);

  printFloat(courseToLondon, gps.location.isValid(), 7, 2);

  const char *cardinalToLondon = TinyGPSPlus::cardinal(courseToLondon);

  printStr(gps.location.isValid() ? cardinalToLondon : "*** ", 6);

  printInt(gps.charsProcessed(), true, 6);
  printInt(gps.sentencesWithFix(), true, 10);
  printInt(gps.failedChecksum(), true, 9);
  Serial.println();

  smartDelay(1000);

  if (millis() > 5000 && gps.charsProcessed() < 10)
    Serial.println(F("No GPS data received: check wiring"));
}

```

valeros, 2021-04-28:
Hi @MNS26 ! It seems that the problem in the framework itself. Could you please try the same project in Arduino IDE?

MNS26, 2021-04-28:
@valeros i tried to compile it with the arduino IDE and this was the output

```
C:\Users\MS26\Desktop\arduino-nightly\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\MS26\Desktop\arduino-nightly\hardware -hardware C:\Users\MS26\AppData\Local\Arduino15\packages -tools C:\Users\MS26\Desktop\arduino-nightly\tools-builder -tools C:\Users\MS26\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\MS26\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\MS26\Desktop\arduino-nightly\libraries -libraries C:\Users\MS26\Documents\Arduino\libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=0000_0000 -ide-version=10814 -build-path C:\Users\MS26\AppData\Local\Temp\arduino_build_817798 -warnings=all -build-cache C:\Users\MS26\AppData\Local\Temp\arduino_cache_490970 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.openocd.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.dfu-util.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\MS26\AppData\Local\Temp\arduino_modified_sketch_480089\Blink.ino
C:\Users\MS26\Desktop\arduino-nightly\arduino-builder -compile -logger=machine -hardware C:\Users\MS26\Desktop\arduino-nightly\hardware -hardware C:\Users\MS26\AppData\Local\Arduino15\packages -tools C:\Users\MS26\Desktop\arduino-nightly\tools-builder -tools C:\Users\MS26\Desktop\arduino-nightly\hardware\tools\avr -tools C:\Users\MS26\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\MS26\Desktop\arduino-nightly\libraries -libraries C:\Users\MS26\Documents\Arduino\libraries -fqbn=arduino:mbed_rp2040:pico -vid-pid=0000_0000 -ide-version=10814 -build-path C:\Users\MS26\AppData\Local\Temp\arduino_build_817798 -warnings=all -build-cache C:\Users\MS26\AppData\Local\Temp\arduino_cache_490970 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.bossac-1.9.1-arduino2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.9.1-arduino2 -prefs=runtime.tools.rp2040tools.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.rp2040tools-1.0.2.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\rp2040tools\1.0.2 -prefs=runtime.tools.openocd.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.openocd-0.11.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\openocd\0.11.0-arduino1 -prefs=runtime.tools.dfu-util.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.dfu-util-0.10.0-arduino1.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\dfu-util\0.10.0-arduino1 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -prefs=runtime.tools.arm-none-eabi-gcc-7-2017q4.path=C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\7-2017q4 -verbose C:\Users\MS26\AppData\Local\Temp\arduino_modified_sketch_480089\Blink.ino
Using board 'pico' from platform in folder: C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.0.0
Using core 'arduino' from platform in folder: C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.0.0
Detecting libraries used...
"C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\7-2017q4/bin/arm-none-eabi-g++" -c -w -g3 -nostdlib "@C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO/defines.txt" "@C:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO/cxxflags.txt" -DARDUINO_ARCH_RP2040 -mcpu=cortex-m0plus -w -x c++ -E -CC -DARDUINO=10814 -DARDUINO_RASPBERRY_PI_PICO -DARDUINO_ARCH_MBED_RP2040 -DARDUINO_ARCH_MBED -DARDUINO_LIBRARY_DISCOVERY_PHASE=0 "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\cores\\arduino" "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\variants\\RASPBERRY_PI_PICO" "-IC:\\Users\\MS26\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\mbed_rp2040\\2.0.0\\co [... cut ...]
C:\Users\MS26\Documents\Arduino\libraries\MPU6050\src\MPU6050.cpp:3329:2: note: in expansion of macro 'printfloatx'
  printfloatx("", Data[2], 5, 0, "\n");
  ^~~~~~~~~~~
Using library TinyGPSPlus-1.0.2b at version 1.0.2 in folder: C:\Users\MS26\Documents\Arduino\libraries\TinyGPSPlus-1.0.2b 
Using library MPU6050 at version 0.2.1 in folder: C:\Users\MS26\Documents\Arduino\libraries\MPU6050 
Using library Wire in folder: C:\Users\MS26\AppData\Local\Arduino15\packages\arduino\hardware\mbed_rp2040\2.0.0\libraries\Wire (legacy)
exit status 1
Error compiling for board Raspberry Pi Pico.

```

MNS26, 2021-04-28:
@valeros Could it be that because the Pi pico supports Serial1 (0 and 1 - 12 and 13 - 16 and 17) and Serial2 (4 and 5 - 8 and 9) on multiple pins?
Compared to how lest say the arduino mega has its Serial ports hard wired to certain pins, could that be part of the problem?
Sadly i dont have enough experience to try and find the problem my self.

valeros, 2021-04-28:
So the same code works as expected in Arduino IDE, right?

MNS26, 2021-04-28:
@valeros nope same result 
[image]

MNS26, 2021-04-28:
the red light on the gps module shows a GPS lock but it still receives no data at all
[image]

valeros, 2021-04-28:
It means that there is an issue in the framework implementation. PlatformIO is not responsible for fixing any issues related to frameworks so I'd recommend you open an issue here https://github.com/arduino/ArduinoCore-mbed

closed 2021-04-28 by valeros