platformio/platform-raspberrypi #4: Add baremetal as framework

MisterSandman7, 2021-05-05:
Although Arduino is very cool it would be nice to be able to write programs in a more classic C/C++ format (especially when tinkering with the C SDK from Raspberry).

LeoDJ, 2021-11-02:
It's already added here: https://github.com/Wiz-IO/wizio-pico
Maybe it could be ported to the official platform?

Crazy-Ginger, 2023-03-25:
The Wizio-pico library has recently disappeared. Tried to get in contact with the dev but no response yet. Would this change be enough to incentivise work on this issue?

jonathanlundstrom, 2023-03-28:
> The Wizio-pico library has recently disappeared. Tried to get in contact with the dev but no response yet. Would this change be enough to incentivise work on this issue?

I've spoken to the dev recently and he will not resume the project. He did not enjoy working with the Pico and the Pi Foundation and therefore decided to cancel it. It's a shame that he went so far as removing it from Github. So I also have the same question. Could this framework be adapted instead?

sstaub, 2023-03-28:
Why not using https://github.com/earlephilhower/arduino-pico ?

jonathanlundstrom, 2023-03-28:
> Why not using https://github.com/earlephilhower/arduino-pico ?

Not bare metal – still Arduino. It's plagued with Wi-Fi and socket issues as well as terrible power consumption. I'm looking for a pure C solution running only the Pico SDK – just like WizIO, but preferably one that still exists.

Spencer-Sawyer, 2023-05-03:
As a note for people looking for the disappeared platform: This commit: https://github.com/blackketter/wizio-pico/commit/a0e41397f445566efca7b26f16af4aa2416aa99b looks like last that was pushed

OzmoOzmo, 2023-05-09:
Hi - I too am looking for a way to use 
-   the Official Pico SDK (Not arduino) 
-   In Windows with Platformio IDE
-   and especially to be able upload and debug.

**The backup of the WizeIO** - would not work with the "Raspberry Pi Pico Debug Probe" (Logs says it is about to upload via CMSIS-DAP but then it goes and makes a .uf2 file and tries to update via flash). Maybe it works with 2 picos (picoprobe) only - but either way, its discontinued and unsupported.

**The earlephilhower** - would not run the most basic print("hello world") - linker errors with the required "stdio_init_all()" call.

All other options Id found seem to be Arduino based - which is not an option for the project I am on.

I got **earlephilhower** to work by copying 6 missing stdio and stdlib files from from the pico sdk to the src folder of my project - so now it can print to platformio's terminal - but anyone have a less hacky way of doing this?

thanks!

sporniket, 2023-06-11:
I am looking forward to this kind of integration too. Thanks @OzmoOzmo  for the feedback about the existing solutions, it helps me a lot in deciding to learn the pico sdk for the time being, as the quickest path to be able to get something.

Because I also looked at how to create a PlatformIO framework, to see if I could do something. And it seems to me that it's not something trivial and quick to do :( I mean, there is a [quick guide](https://docs.platformio.org/en/latest/platforms/creating_platform.html) but for me, going further (even with starting from the two existing solutions) require : 
* being proficient with the pico sdk and CMake, to know how the standard sdk works. (to be done, I was planning to just use platformIO on VSCode, as usual)
* being proficient with platformIO and SCons, to know how platformio works (to be done, I was planning to just use the upload button on VSCode and the platformio.ini file, as usual)
* study in depth one or two existing and mature non-arduino platform/frameworks, e.g. like ESP32 (to be done, I was expecting the integration to already exist, as usual)

jonathanlundstrom, 2023-06-11:
When wizio-pico was taken down I started working on a forked version that removes the Arduino Framework. The goal was to clean up the code and get an understanding of the previous version of the SDK (severely rearranged) was included, to see if I could include the new one as a submodule and get it working.

Unfortunately I quickly realized that I need to learn more about PlatformIO and the build/debug tools to be able to implement something properly, and this is something that I don't have time for at the moment - unfortunately 🥺

I've pushed the code here if anybody else wants to take a look at it, or base something on it: https://github.com/ec-solutions/ez-pico. The README is still from the original wizio-pico project.

It would be fun if we could work on something together.

sporniket, 2023-06-12:
If I ever have time to study all that stuff, I'll give it a try. Don't hold your breath, though.

whatisbyandby, 2023-06-27:
> When wizio-pico was taken down I started working on a forked version that removes the Arduino Framework. The goal was to clean up the code and get an understanding of the previous version of the SDK [...]

@jonathanlundstrom I would be interested in contributing to this. I stumbled on this issue, because I am interested in the same thing, programming the pico using the C/C++ SDK instead of all the overhead of using the Arduino framework. I'm working on an MQTT Smart Thermostat with a Pico-W using the SDK. I'm finding it pretty clunky to run UTs on my dev machine, and then switch to target the pico. Seems like Platformio would be a good solution to this problem, and I was debating tweaking the code to work with Arduino so I could use it, but I would be much happier with using the SDK with Platformio.

KenwoodFox, 2023-11-27:
Anecdotal but in the meantime, ive been experimenting with https://github.com/maxgerhardt/wizio-pico.git and it seems to work for me (but i have limited experience)

I would love to see a new baremetal framework that just uses https://github.com/raspberrypi/pico-sdk or maybe another alternative to just include the pico-sdk

AgainPsychoX, 2024-03-20:
Any news for official support?

Regeneric, 2024-04-29:
Is there anything new in this matter?

sstaub, 2024-04-29:
There is a new extension for vscode from the vendor https://marketplace.visualstudio.com/items?itemName=raspberry-pi.raspberry-pi-pico

KenwoodFox, 2024-05-08:
Thats pretty cool but dosn't fix the issues in platformio right?

KenwoodFox, 2024-05-09:
Thats a real shame, what about using PIO to just handle building the sorta, more raspberrypi specific cmake stuff? that could still be handy even if its just backpacking on the existing SDK

sstaub, 2024-05-09:
There was a big discussion about here https://github.com/platformio/platform-raspberrypi/pull/36
The PIO team wants money from the raspberry pi foundation.

OzmoOzmo, 2024-05-09:
Hi, That link of the spat with pio and rpi is very disappointing.
Ive have Earlephilhower working perfectly for my needs. PIO compiles (as far as i remember)
Baremetal pico with no need use Arduino framework but I can serial.print() from pico back to the platformio console. 
Step Debugging works perfectly using the official pico debugger. And all uploads upload very fast without having reset the pico.  All the pico sdk examples work ok. Setup is as I described above.
If this is useful, if anyone wants to try replicate what I have - to help document the setup for others, let me know.

zbauman3, 2024-07-14:
Just adding in here that I'd love `pico-sdk` support in PlatformIO. It's a shame that we have to switch between different dev environments for different MCUs.

PlatformIO is really a game changer for development, but not supporting one of the most popular MCUs is _very_ disappointing.

sporniket, 2024-07-14:
Yep, so disappointing. And a wake up call that some parts of PlatformIO should not be taken for granted. So much so that I plan to learn how to operate without PlatformIO, "just in case", even if I still plan to use it as a primary dev platform for supported platforms and toolchains.

Ironman-007, 2024-10-08:
I tried this platformio.ini and it works:

[env:stable]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = picosdk
board_build.core = earlephilhower

debug_tool = jlink
upload_protocol = jlink

One problem is I haven't tried the PIO, so not sure if it supports PIO. I also just started so not sure if this just has limited functions of the PicoSDK.

znmeb, 2024-10-08:
The `earlephilhower` core aka Arduino-Pico (https://github.com/earlephilhower/arduino-pico) does connect with the Raspberry Pi Pico SDK. See https://arduino-pico.readthedocs.io/en/latest/sdk.html for the documentation. Is a "baremetal framework" necessary given that capability?

OzmoOzmo, 2024-10-08:
earlephilhower couldnt print back to the platformio terminal last time i tried it… Maybe its been fixed?