platformio/platform-raspberrypi #5: Request to Add J-Link and ST-Link Upload & Debug Support to RP2040 Pico

key1066, 2021-05-06:
Request to add Currently, PlatformIO only seems to support `cmsis-dap and raspberry-swd` debug options. `jlink` and `stlink` do not appear to be supported.

SEGGER have updated their J-Link probe firmware to support Pico uploads (flash to QSPI) and debugging, see  https://www.segger.com/products/debug-probes/j-link/ for `Release Notes`.

I've been able to debug a Pico using a combination of `VSCODE + cmake +Picoprobe` and the `PlatformIO Debugger`. In addition, a Pico `elf` can be debugged using `SEGGER Educ Mini probe & Ozone Debugger`.

I'm not sure what changes are required in PlatformIO Debugger to work but the Raspberry Pico probe needed `OpenOCD` to be built with the Picoprobe driver, as detailed in the Raspberry Pi documentation at  https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf.

It would be highly productive if we can have out-of-the-box J-Link (and ST-Link) support for RP2040 Pico uploads & debugging.

zpm1066, 2021-05-06:
Raspberry Pi Picoprobe seems to be using a forked branch of OpenOCD with jlink probe.
 https://github.com/raspberrypi/openocd/tree/rp2040_jlink

platformIO may have to do something similar.

valeros, 2021-05-07:
Hi @zpm1066 ! J-Link should be available in the dev branch. Would be great if you could try the following configuration for both uploading and debugging:
```ini
[env:pico]
platform = raspberrypi
framework = arduino
board = pico

upload_protocol = jlink
debug_tool = jlink
```

As for ST-Link, it seems that it's not compatible with RP2040, at least now https://github.com/raspberrypi/openocd/issues/13

zpm1066, 2021-05-07:
Hi @valeros,

PlatformIO sees `jlink` as unknown debug tool. So, I tried a few different things.

```
Environment: macOS 11.3.1, VSCODE: 1.56.0, PlatformIO 2.3.2
Debug Probe: Segger Edu Mini with the latest firmware
```

Previously, I've successfully used the above `Edu Mini` to debug a Pico using `VSCODE + Picoprobe + PlatformIO Debugger.`

Would it be possible to provide a version of PlatformIO that recognizes `jlink` as a valid debug tool?

Regarding the  error `Writing target memory failed` in Method #2, I'll raise it with the SEGGER online Support forum to see if they can assist.

I'm available for further testing as required. Thank you.

Method #1 
----------

I used the following `platformio.ini`, as per your suggestion. 

```
[env:pico]
platform = raspberrypi
framework = arduino
board = pico
upload_protocol = jlink
debug_tool = jlink
debug_init_break = tbreak main
```

I got the following output, with error `Unknown debug tool jlink` and unable to build successfully.

```
> Executing task in folder tiny2040_blink: platformio run <

Processing pico (platform: raspberrypi; framework: arduino; board: pico)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
**DebugInvalidOptionsError: Unknown debug tool `jlink`. Please use one of `cmsis-dap, raspberrypi-swd` or `custom`:**
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/builder/main.py", line 177:
    env.SConscript("$BUILD_SCRIPT")
  File "/Users/quark11/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 591:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/quark11/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Script/SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/quark11/.platformio/platforms/raspberrypi/builder/main.py", line 121:
    target_elf = env.BuildProgram()
  File "/Users/quark11/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 61:
    env.ProcessProgramDeps()
  File "/Users/quark11/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/platformio.py", line 113:
    env.PrintConfiguration()
  File "/Users/quark11/.platformio/packages/tool-scons/scons-local-4.1.0/SCons/Util.py", line 658:
    return self.method(*nargs, **kwargs)
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/pioplatform.py", line 234:
    _get_debug_data(),
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/builder/tools/pioplatform.py", line 198:
    % board_config.get_debug_tool_name(env.GetProjectOption("debug_tool")),
  File "/Users/quark11/.platformio/penv/lib/python3.9/site-packages/platformio/platform/board.py", line 139:
    raise DebugInvalidOptionsError(
```

Method #2 
-----------

[... 81 lines cut ...]
PC = 000020F0, CycleCnt = 00000000
R0 = 501008B0, R1 = 00000000, R2 = 00003ECC, R3 = 00000000
R4 = 50100E84, R5 = 00000000, R6 = 00000000, R7 = 50100EAC
R8 = FFFFFFFF, R9 = FFFFFFFF, R10= FFFFFFFF, R11= FFFFFFFF
R12= 50100000
SP(R13)= 50100898, MSP= 50100898, PSP= FFFFFFFC, R14(LR) = 000002B3
XPSR = 61000000: APSR = nZCvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 00000000, CONTROL = 00, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.

Downloading file [.pio/build/pico/firmware.bin]...
Writing target memory failed.

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Script processing completed.

========================================================== [SUCCESS] Took 1.60 seconds ==========================================================

Terminal will be reused by tasks, press any key to close it.

```

valeros, 2021-05-07:
My bad, sorry, you need this config:
```ini
[env:pico]
platform = https://github.com/platformio/platform-raspberrypi.git
framework = arduino
board = pico

upload_protocol = jlink
debug_tool = jlink
```

valeros, 2021-05-07:
Hmm, is there any chance you have two instances of the `raspberrypi` platform installed? What is the output of `pio platform show raspberrypi`?

zpm1066, 2021-05-07:
Here's the requested output.

```
quark11:tiny2040_blink quark11$ pio platform show raspberrypi
raspberrypi ~ Raspberry Pi RP2040
=================================
RP2040 is a low-cost, high-performance microcontroller device with with a large on-chip memory, symmetric dual-core processor complex, and rich peripheral.

Version: 1.0.0+sha.6916058
Home: https://www.raspberrypi.org/documentation/rp2040/getting-started/
Repository: https://github.com/platformio/platform-raspberrypi.git
Vendor: https://www.raspberrypi.org/documentation/rp2040/getting-started/
License: Apache-2.0
Frameworks: arduino

Packages
--------

Package toolchain-gccarmnoneeabi
--------------------------------
Type: toolchain
Requirements: ~1.90201.0
Installed: Yes
Version: 1.90201.191206
Original version: 9.2.1
Description: GNU toolchain for Arm Cortex-M and Cortex-R processors

Package framework-arduino-mbed
------------------------------
Type: framework
Requirements: ~2.0.0
Installed: Yes
Version: 2.0.0
Original version: None
Description: Arduino framework supporting mbed-enabled boards

Package tool-rp2040tools
------------------------
Type: uploader
Requirements: ~1.0.2
Installed: Yes
Version: 1.0.2
Original version: None
Description: Tools for interacting with a RP2040 device in BOOTSEL mode or with a RP2040 binary

Package tool-openocd-raspberrypi
--------------------------------
Type: uploader
Requirements: ~2.1100.0
Installed: Yes
Version: 2.1100.0
Original version: 11.0
Description: Open On-Chip Debugger for Raspberry Pi MCUs

Package tool-jlink
------------------
Type: uploader
Requirements: ~1.72000.0
Installed: Yes
Version: 1.72000.0
Original version: 7.20.0
Description: Software and Documentation Pack for SEGGER J-Link debug probes

Boards
------
ID                 MCU     Frequency    Flash    RAM    Name
-----------------  ------  -----------  -------  -----  ---------------------------
nanorp2040connect  RP2040  133MHz       2MB      264KB  Arduino Nano RP2040 Connect
pico               RP2040  133MHz       2MB      264KB  Raspberry Pi Pico

```

valeros, 2021-05-07:
As I thought, you have the dev platform installed from this repository, but it's outdated. Please run `pio platform update raspberrypi` and try again.

zpm1066, 2021-05-07:
OK. I've updated the platform. 

Here's the output:

```
> Executing task in folder tiny2040_blink: platformio run --target upload <

Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
-------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.0.0+sha.5271d82) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (jlink) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - tool-jlink 1.72000.0 (7.20.0) 
 - tool-openocd-raspberrypi 2.1100.0 (11.0) 
 - tool-rp2040tools 1.0.2 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/pico/src/main.cpp.o
Compiling .pio/build/pico/FrameworkArduinoVariant/double_tap_usb_boot.c.o
Compiling .pio/build/pico/FrameworkArduinoVariant/variant.cpp.o
Compiling .pio/build/pico/FrameworkArduino/Interrupts.cpp.o
Compiling .pio/build/pico/FrameworkArduino/Serial.cpp.o
Compiling .pio/build/pico/FrameworkArduino/Tone.cpp.o
Archiving .pio/build/pico/libFrameworkArduinoVariant.a
Compiling .pio/build/pico/FrameworkArduino/USB/PluggableUSBDevice.cpp.o
Indexing .pio/build/pico/libFrameworkArduinoVariant.a
Compiling .pio/build/pico/FrameworkArduino/USB/USBCDC.cpp.o
Compiling .pio/build/pico/FrameworkArduino/USB/USBSerial.cpp.o
Compiling .pio/build/pico/FrameworkArduino/WMath.cpp.o
Compiling .pio/build/pico/FrameworkArduino/abi.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/Common.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/IPAddress.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/PluggableUSB.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/Print.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/Stream.cpp.o
Compiling .pio/build/pico/FrameworkArduino/api/String.cpp.o
Compiling .pio/build/pico/FrameworkArduino/itoa.c.o
Compiling .pio/build/pico/FrameworkArduino/main.cpp.o
Compiling .pio/build/pico/FrameworkArduino/mbed/platform/cxxsupport/mstd_mutex.cpp.o
Compiling .pio/build/pico/FrameworkArduino/pinToIndex.cpp.o
Compiling .pio/build/pico/FrameworkArduino/wiring.cpp.o
Compiling .pio/build/pico/FrameworkArduino/wiring_analog.cpp.o
Compiling .pio/build/pico/FrameworkArduino/wiring_digital.cpp.o
Compiling .pio/build/pico/FrameworkArduino/wiring_pulse.cpp.o
Compiling .pio/build/pico/FrameworkArduino/wiring_shift.cpp.o
Archiving .pio/build/pico/libFrameworkArduino.a
Indexing .pio/build/pico/libFrameworkArduino.a
Linking .pio/build/pico/firmware.elf
Generating UF2 image
/Users/quark11/.platformio/packages/tool-rp2040tools/elf2uf2 ".pio/build/pico/firmware.elf" ".pio/build/pico/firmware.uf2"
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 50500 bytes from 270336 bytes)
Flash: [          ]   3.5% (used 73320 bytes from 2097152 bytes)
Building .pio/build/pico/firmware.bin
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = jlink
Uploading .pio/build/pico/firmware.bin
SEGGER J-Link Commander V7.20 (Compiled Apr 28 2021 17:34:16)
DLL version V7.20, compiled Apr 28 2021 17:34:09

J-Link Command File read successfully.
Processing script file...

J-Link connection not established yet but required for command.
Connecting to J-Link via USB...O.K.
Firmware: J-Link EDU Mini V1 compiled Feb 18 2021 11:25:23
Hardware version: V1.00

License(s): FlashBP, GDB
VTref=3.297V
Target connection not established yet but required for command.
Device "RP2040_M0_0" selected.

Connecting to target via SWD
ConfigTargetSettings() start
J-Link script: ConfigTargetSettings()
ConfigTargetSettings() end
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x0BC12477
DPIDR: 0x0BC12477
Scanning AP map to find all available APs
AP[1]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x04770031)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410CC601. Implementer code: 0x41 (ARM)
Found Cortex-M0 r0p1, Little endian.
FPUnit: 4 code (BP) slots and 0 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
ROMTbl[0][0]: E000E000, CID: B105E00D, PID: 000BB008 SCS
ROMTbl[0][1]: E0001000, CID: B105E00D, PID: 000BB00A DWT
ROMTbl[0][2]: E0002000, CID: B105E00D, PID: 000BB00B FPB
Cortex-M0 identified.
PC = 10009B50, CycleCnt = 00000000
R0 = 0003D090, R1 = 01734736, R2 = 40054000, R3 = 0000BD21
R4 = 0003D090, R5 = 00000000, R6 = 00000003, R7 = 00000000
R8 = 00000000, R9 = 00000000, R10= 00000000, R11= 00000000
R12= 000000FA
SP(R13)= 2000AD00, MSP= 2003FFC0, PSP= 2000AD00, R14(LR) = 100092C7
XPSR = 81000000: APSR = Nzcvq, EPSR = 01000000, IPSR = 000 (NoException)
CFBP = 02000000, CONTROL = 02, FAULTMASK = 00, BASEPRI = 00, PRIMASK = 00
FPU regs: FPU not enabled / not implemented on connected CPU.

**Downloading file [.pio/build/pico/firmware.bin]...
Writing target memory failed.**

Reset delay: 0 ms
Reset type NORMAL: Resets core & peripherals via SYSRESETREQ & VECTRESET bit.
Reset: Halt core after reset via DEMCR.VC_CORERESET.
Reset: Reset device via AIRCR.SYSRESETREQ.

Script processing completed.

========================================================= [SUCCESS] Took 11.07 seconds =========================================================

Terminal will be reused by tasks, press any key to close it.
```

valeros, 2021-05-07:
Debugging doesn't work as well, right?

zpm1066, 2021-05-07:
Good news! Debug seems to work.

zpm1066, 2021-05-07:
I'm able to single step through the blink sketch (RGB version of blink) on my `Pimoroni Tiny2040` which a RGB neopixel.

```
Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.0.0+sha.5271d82) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (jlink) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES: 
 - framework-arduino-mbed 2.0.0 
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 26 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Checking size .pio/build/pico/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 50500 bytes from 270336 bytes)
Flash: [          ]   3.6% (used 75192 bytes from 2097152 bytes)
========================= [SUCCESS] Took 1.06 seconds =========================
SEGGER J-Link GDB Server V7.20 Command Line Version

JLinkARM.dll V7.20 (DLL compiled Apr 28 2021 17:34:09)

Command line: -singlerun -if SWD -select USB -device RP2040_M0_0 -port 2331
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      yes
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 RP2040_M0_0
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link EDU Mini V1 compiled Feb 18 2021 11:25:23
Hardware: V1.00
S/N: 801022671
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...
Reading symbols from /Users/quark11/Documents/PlatformIO/CompletedProjects/pico_arduino/tiny2040_blink/.pio/build/pico/firmware.elf...
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x10009B50 (Data = 0xBD10D3FB)
Read 2 bytes @ address 0x10009B50 (Data = 0xD3FB)
0x10009b50 in busy_wait_us_32 ()
Reading 64 bytes @ address 0x10009B40
Received monitor command: clrbp
Received monitor command: speed auto
Select auto target interface speed (2000 kHz)
Select auto target interface speed (2000 kHz)
Received monitor command: reset
Resetting target
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x000000EE)
Loading section .boot2, size 0x100 lma 0x10000000
Downloading 256 bytes @ address 0x10000000
Loading section .text, size 0xd6e8 lma 0x10000100
Downloading 15984 bytes @ address 0x10000100
Downloading 15920 bytes @ address 0x10003F70
Downloading 15920 bytes @ address 0x10007DA0
Downloading 7192 bytes @ address 0x1000BBD0
Loading section .rodata, size 0x225c lma 0x1000d7e8
Downloading 8796 bytes @ address 0x1000D7E8
Loading section .ARM.exidx, size 0x8 lma 0x1000fa44
Downloading 8 bytes @ address 0x1000FA44
Loading section .data, size 0x2c6c lma 0x1000fa4c
Downloading 11372 bytes @ address 0x1000FA4C
Loading section .uninitialized_ram.magic_location, size 0xc lma 0x100126b8
Downloading 12 bytes @ address 0x100126B8
Start address 0x100001d4, load size 75460
Writing register (PC = 0x100001d4)
Transfer rate: 9211 KB/sec, 8384 bytes/write.
Read 4 bytes @ address 0x100001D4 (Data = 0x49202000)
Read 2 bytes @ address 0x100001D4 (Data = 0x2000)
Reading 64 bytes @ address 0x10002380
Read 2 bytes @ address 0x100023AA (Data = 0xF000)
Temporary breakpoint 1 at 0x100023aa: file /Users/quark11/.platformio/packages/framework-arduino-mbed/cores/arduino/main.cpp, line 36.
PlatformIO: Initialization completed
Reading 64 bytes @ address 0x10000300
Read 2 bytes @ address 0x1000032C (Data = 0x2101)
Reading 64 bytes @ address 0x10000340
Read 2 bytes @ address 0x10000360 (Data = 0x2103)
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Setting breakpoint @ address 0x1000032C, Size = 2, BPHandle = 0x0001
Setting breakpoint @ address 0x10000360, Size = 2, BPHandle = 0x0002
Setting breakpoint @ address 0x100023AA, Size = 2, BPHandle = 0x0003
Starting target CPU...
...Breakpoint reached @ address 0x1000032C
Reading all registers
Removing breakpoint @ address 0x1000032C, Size = 2
Removing breakpoint @ address 0x10000360, Size = 2
Removing breakpoint @ address 0x100023AA, Size = 2
Read 4 bytes @ address 0x1000032C (Data = 0x20132101)
Reading 64 bytes @ address 0x20041F00
WARNING: Failed to read memory @ address 0x60984388

Breakpoint 2, blink (led=-1, n=-1) at src/main.cpp:19
19	        digitalWrite(LED_GREEN_PIN, HIGH);
WARNING: Failed to read memory @ address 0x60984388

```

zpm1066, 2021-05-07:
Odd that the release upload doesn't work but debug look fine so far.

Here are a few photos.

[image]

[image]

vorosj, 2021-05-09:
Is it possible to use the Segger tools instead of the openocd? Something like this:
debug_server =
  /full/path/to/JLinkGDBServerCL
And a custom upload protocol pointing to J-Flash Lite?

zpm1066, 2021-05-09:
I'll look into your suggestion. I did try flashing the `firmware.bin` file using `JFlashLite` but it threw up the same error,

```
Downloading file [.pio/build/pico/firmware.bin]...
Writing target memory failed.
```

If the `bin` file is ok,  `JFlashLite` normally works. It is possible that there is a bug  in SEGGER `jlink` firmware to do with flashing. Previously, using a combination of `VSCODE + cmake + PlatformIO`, I've been able to debug using `jlink` but I'm not aware of anyone succeeding an `upload`. 

SEGGER added write to `QSPI` but I'll follow up the debug output with SEGGER support.

vorosj, 2021-05-09:
.bin has no start address, and default 0 is incorrect, try 0x10000000.

zpm1066, 2021-05-09:
The .bin, .uf2, and .elf files generated by PlatformIO work fine. 

Code upload Pico using `JFlashLite with Edu Mini J-Link` works,  and also .uf2 and .elf files work correctly.

`JFlashLite` correctly uses the start address of `0x10000000`.

How do I force PlatformIO upload to use  address of `0x10000000`?

vorosj, 2021-05-09:
Hi zpm1066,

I studied the Segger docs, and it seems the JflashLite is not suitable for the job. JFlash is capable, but not working with the edu tools. I think JLink.exe can do it as well, but I didn't tried yet. It needs some scripting.  
https://wiki.segger.com/J-Link_Commander
If I understand well, the following commands required:
device <DeviceName>
loadfile <Filename> [<Addr>]

> Can we get PlatformIO to generate .hex file as well, like with other platforms (e.g. nRF52840), then we won't need to be concerned with the start address? Thanks.

A Python script for the job: https://www.programmersought.com/article/38735814995/

zpm1066, 2021-05-09:
Hi @vorosj,

Thanks. I wonder whether PlatformIO can adapt what is being used by platform/platform-nordicnrf52 for .hex files? 

The PlatformIO developers may know more.

zpm1066, 2021-05-09:
Hi @valeros, @ivankravets,

We're so close to getting both uploads & debug to work for Pico with J-Link.

Can we get PlatformIO modified to use .hex files for RP2040 J-link uploads? Thanks.

valeros, 2021-05-10:
@vorosj Do you mind sharing the entire log from the debug console?

vorosj, 2021-05-10:
If I use Ozone with the.elf, the following happens:
- Ozone shows the main.cpp .../arduino version, so I can't view my program code
- otherwise debugging seems to work, stops on breakpoints. led is blinking when the program is running etc.

vorosj, 2021-05-10:
_Do you mind sharing the entire log from the debug console?_

It's on the console after star the debug session, it stopped at init();
Processing pico (platform: https://github.com/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.0.0+sha.079d5f4) > Raspberry Pi Pico
HARDWARE: RP2040 133MHz, 264KB RAM, 2MB Flash
DEBUG: Current (jlink) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed 2.0.0
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 27 compatible libraries
Scanning dependencies...
No dependencies
Building in debug mode
Checking size .pio\build\pico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.7% (used 50500 bytes from 270336 bytes)
Flash: [          ]   3.6% (used 74888 bytes from 2097152 bytes)
========================= [SUCCESS] Took 1.22 seconds =========================
SEGGER J-Link GDB Server V7.20 Command Line Version

JLinkARM.dll V7.20 (DLL compiled Apr 28 2021 17:34:08)

Command line: -singlerun -if SWD -select USB -device RP2040_M0_0 -port 2331
-----GDB Server start settings-----
GDBInit file:                  none
GDB Server Listening port:     2331
SWO raw output listening port: 2332
Terminal I/O port:             2333
Accept remote connection:      localhost only
Generate logfile:              off
Verify download:               off
Init regs on start:            off
Silent mode:                   off
Single run mode:               on
Target connection timeout:     0 ms
------J-Link related settings------
J-Link Host interface:         USB
J-Link script:                 none
J-Link settings file:          none
------Target related settings------
Target device:                 RP2040_M0_0
Target interface:              SWD
Target interface speed:        4000kHz
Target endian:                 little

Connecting to J-Link...
J-Link is connected.
Firmware: J-Link V11 compiled Apr 27 2021 16:36:21
Hardware: V11.00
S/N: 261006780
OEM: SEGGER-EDU
Feature(s): FlashBP, GDB
Checking target voltage...
Target voltage: 3.30 V
Listening on TCP/IP port 2331
Connecting to target...
Connected to target
Waiting for GDB connection...
Reading symbols from c:\Users\mmormota\OneDrive\Documents\PlatformIO\Projects\pico_ard1\.pio\build\pico\firmware.elf...
undefinedC:\Users\mmormota\.platformio\packages\toolchain-gccarmnoneeabi\bin\arm-none-eabi-gdb.exe: warning: Couldn't determine a path for the index cache directory.
PlatformIO Unified Debugger -> http://bit.ly/pio-debug
PlatformIO: debug_tool = jlink
PlatformIO: Initializing remote target...
Connected to 127.0.0.1
Reading all registers
Read 4 bytes @ address 0x10007420 (Data = 0x223FD104)
Read 2 bytes @ address 0x10007420 (Data = 0xD104)
Received monitor command: clrbp
Received monitor command: speed auto
Select auto target interface speed (2000 kHz)
Received monitor command: reset
Resetting target
Received monitor command: halt
Halting target CPU...
...Target halted (PC = 0x000000EE)
Downloading 256 bytes @ address 0x10000000
Downloading 15968 bytes @ address 0x10000100
Downloading 15936 bytes @ address 0x10003F60
Downloading 15920 bytes @ address 0x10007DA0
Downloading 6904 bytes @ address 0x1000BBD0
Downloading 8780 bytes @ address 0x1000D6C8
Downloading 8 bytes @ address 0x1000F914
Downloading 11372 bytes @ address 0x1000F91C
Downloading 12 bytes @ address 0x10012588
0x10007420 in mbed::FilePath::FilePath(char const*) ()
Select auto target interface speed (2000 kHz)
Resetting target
Loading section .boot2, size 0x100 lma 0x10000000
Loading section .text, size 0xd5c8 lma 0x10000100
Loading section .rodata, size 0x224c lma 0x1000d6c8
Loading section .ARM.exidx, size 0x8 lma 0x1000f914
Loading section .data, size 0x2c6c lma 0x1000f91c
Loading section .uninitialized_ram.magic_location, size 0xc lma 0x10012588
Start address 0x100001d4, load size 75156
Transfer rate: 36697 KB/sec, 8350 bytes/write.
Temporary breakpoint 1 at 0x100022f6: file C:\Users\mmormota\.platformio\packages\framework-arduino-mbed\cores\arduino\main.cpp, line 36.
PlatformIO: Initialization completed
Writing register (PC = 0x100001d4)
Read 4 bytes @ address 0x100001D4 (Data = 0x49202000)
Read 2 bytes @ address 0x100001D4 (Data = 0x2000)
Reading 64 bytes @ address 0x100022C0
Read 2 bytes @ address 0x100022F6 (Data = 0xF000)
PlatformIO: Resume the execution to `debug_init_break = tbreak main`
PlatformIO: More configuration options -> http://bit.ly/pio-debug
Setting breakpoint @ address 0x100022F6, Size = 2, BPHandle = 0x0001
Starting target CPU...
...Breakpoint reached @ address 0x100022F6
Reading all registers
Removing breakpoint @ address 0x100022F6, Size = 2
Read 4 bytes @ address 0x100022F6 (Data = 0xF857F000)

Temporary breakpoint 1, main () at C:\Users\mmormota\.platformio\packages\framework-arduino-mbed\cores\arduino\main.cpp:36
36		init();

vorosj, 2021-05-10:
As it doesn't stop after the single step, I pushed pause:

Program
 received signal SIGTRAP, Trace/breakpoint trap.
0x100105ec in ?? ()
Debugger requested to halt target...
...Target halted (PC = 0x100105EC)
Reading all registers
Removing breakpoint @ address 0x100022FA, Size = 2
Read 4 bytes @ address 0x100105EC (Data = 0x432108F9)
Read 2 bytes @ address 0x100105EC (Data = 0x08F9)
Read 4 bytes @ address 0x10010C6C (Data = 0x00020001)
Read 2 bytes @ address 0x10010C6C (Data = 0x0001)
Read 4 bytes @ address 0x10010C6C (Data = 0x00020001)
Read 2 bytes @ address 0x10010C6C (Data = 0x0001)

vorosj, 2021-05-10:
The Led is not blinking if I push continue.
(but blinks if I stop debug session and push the hw reset button on the target)

vorosj, 2021-05-10:
This is the test source:

#include <Arduino.h>

void setup() {
  // put your setup code here, to run once:
  pinMode(PICO_DEFAULT_LED_PIN, OUTPUT);
}

void loop() {
  // put your main code here, to run repeatedly:
  digitalWrite(PICO_DEFAULT_LED_PIN, HIGH);
  delay(1000);
  digitalWrite(PICO_DEFAULT_LED_PIN, LOW);
  delay(1000);
}

zpm1066, 2021-05-10:
Hi @ valeros, @vorosj,

Thank you very much. I've updated the `raspberrypi` platform and can both upload & debug a simple RGB blink program on my `Pimoroni Tiny2040` board with a `Edu Mini J-Link `probe..

I can single step through the code but the debug behavior is not always consistent. The debugger doesn't hit a breakpoint in my blink's loop() until a hit restart from within the PIO debugger after the first time I enter the debugger. Subsequent step into & out and stopovers are fine. 

So it seems there may be a minor quirk with the debugging but I'll do some more debugging shortly and report back.

The upload of the .hex file works consistently. I have tested it on several of my Mac systems - macOS 11.3.1 (MacBook Pro 13"), macOS 14.6 MacBook Air Early 2014 mid-2014) and macOS 10.12.6 (Mac mini mid-2011).

@ valeros,
Very much appreciate the switch to the .hex file upload and your continuing efforts. Thank you.

@ vorosj,
I'm not sure why your blink LED doesn't work. 

Try the following:
1. Set your break points in your blink program, say in setup() and loop()
2. Run debug build
3. From the PIO debugger tool bar, hit the Restart button again
4. Hit Continue (F5)
5. You should reach your first breakpoint
6. Try single step over or into & out

The above works for me consistently in debugging my RGB blink program.

vorosj, 2021-05-10:
Thank you very much, it's working. :-)

zpm1066, 2021-05-10:
Hi @vorosj, 

Wonderful! 

I'm not quite sure why a RESET is required but we now have a one-stop solution for Pico development with PlatformIO.

I have multiple Pico boards - official Raspberry Pi Pico board, Pimoroni Tiny2040, and Adafruit RP2040. That latter has the LiPo battery support and option to add a standard J-link connector but I haven't used it much yet.

The Pimoroni Tiny2040 is a great little Pico board with SWD and 8GB Flash. Works fine with ILI9341, ST7789 and SSD1306 displays. It's my preferred Pico board at the moment.

zpm1066, 2021-05-10:
Thanks @vorosj. I'll review and reproduce your debug findings.

I hope to do some more Pico debugging later today.

vorosj, 2021-05-10:
Usually I am working with bare metal, but Arduino is great because of the sensor, lcd, Wifi etc libraries. Using an ILI9341 LCD is really comfortable in Arduino. I like Arduino because of PlatformIO, before that I tried the Arduino IDE and concluded that it's unuseful for anything bigger than Led blinking... :-)
Not sure what Mbed is doing in the Arduino platform, is it a real-time operating system? 
For bare metal development my preferred IDE is Keil, but there is no support for the Pico yet. PlatformIO is working with the Wizio, with Ozone debugging, or I can use the  Segger IDE too.

CesiumK9, 2021-06-20:
Morning all... :)

Just this past week I acquired a Pi Pico board and a EDU mini, I installed support for them in PlatformIO and update the firmware for the EDU mini.

I first uploaded the blink LED code via 'picotool binary' method repeatedly with no problem, then tried via J-Link then the problem occurred. The first upload of blink hex is successful, on the second upload of the exact same hex file the EDU mini reports 'ERROR: Could not connect to target'.

I did the same with 'J-Flash Lite V7.20' program with exact same LED hex file compiled by platformio to directly upload to the Pi Pico board bypassing platformio completely… 

This is my link to my original platformio community post, I can re-post necessary bits of it as required...
https://community.platformio.org/t/raspberry-pi-pico-j-link-problem/22062/4

[image]

zpm1066, 2021-06-20:
Power down the Raspberry Pi Pico and then flash it. 

I can't recall if a power down was needed before for J-Link but it seems now that with PlatformIO or `J-Flash Lite`, a Pico reset may not be enough. The power down method works consistently.

I tried to replicate your test with `J-Flash Lite`, as below. 
The `ERROR: Could not connect to target.`  appeared when I tried a Pico reset. Successful download after a power down & reconnect.

[image]

zpm1066, 2021-06-20:
Hi @valeros,

It seems now that in order to flash a Raspberry Pi Pico with J-Link, a power cycle of the Pico is required. I don't think that was the case earlier back in May.

Are you seeing the same? If you can, please confirm. 

We should be able to flash the Pico without a power cycle or a reset of the board. 

Thank you.

zpm1066, 2021-07-07:
The above issue has been fixed by Segger with J-Link firmware v7.5 update.

Please update PlatformIO with J-Link v7.5 files. Thanks.

closed 2021-07-07 by zpm1066

EmBitz, 2021-09-23:
For the STlink you could use Raspberry version EBlink 

https://github.com/EmBitz/EBlink/tree/master/packages