platformio/platform-raspberrypi #6: -DF_CPU missing

blackketter, 2021-05-12:
I believe that `build_flags` should include  `-DF_CPU=133000000` by default for the RP2040 boards.  

I see that "f_cpu" is defined in the pico.json file, but that doesn't seem to propagate down to the compiler flags.

valeros, 2021-05-13:
Hi @blackketter ! There is no such  macro in the official repo https://github.com/arduino/ArduinoCore-mbed/blob/master/variants/RASPBERRY_PI_PICO/defines.txt

closed 2021-05-13 by valeros

blackketter, 2021-05-13:
Thanks, @valeros !  So would this be a bug against the ArduinoCore-mbed?  It seems to be defined for every other Arduino board I've checked...

valeros, 2021-05-13:
> So would this be a bug against the ArduinoCore-mbed?

I think so, but that macro doesn't seem to be very widely used with Arm-based boards.