platformio/platform-raspberrypi #8: tool-rp2040tools seems to be a required dependency instead of optional (elf2uf2 is required during build)

fsievers, 2021-05-26:
When using platform-raspberrypi in a CI build with GitLab-CI (this would be the case with other CIs), the build with `pio run …` or `pio ci …` is unable to build the uf2 firmware file. The problem is, that `elf2uf2` is not found because it's part of the optional dependency `tool-rp2040tools`.

Here is a minimal example `.gitlab-ci.yml`:
```
image: python:3.8

stages:
  - build

before_script:
  - "pip install -U platformio"

job:
  stage: build
  script: "pio run -e pico"
```

In the build stage when running `pio run -e pico` (with `env:pico` using `platform=raspberrypi`), the output from the CI is as follow:
```
…
Linking .pio/build/raspberrypi/firmware.elf
Generating UF2 image
elf2uf2 ".pio/build/raspberrypi/firmware.elf" ".pio/build/raspberrypi/firmware.uf2"
sh: 1: elf2uf2: not found
*** Error 127
Checking size .pio/build/raspberrypi/firmware.elf
…
```

One way to workaround this is to change the `before_script` to the following:
```
before_script:
  - "pip install -U platformio"
  - "pio platform install https://github.com/platformio/platform-raspberrypi.git --with-package=uploader" # we need rp2040tools to get elf2uf2
```

The problem is, that the `pio run …` command does not install optional dependencies, but the `tool-rp2040tools` seems to be a required dependency for `platform-raspberrypi`.

valeros, 2021-05-26:
Thanks for pointing it out, should be fixed in the dev branch.

closed 2021-05-26 by valeros

fsievers, 2021-05-26:
Thank you for your fast response. I can confirm that the problem is fixed in the development branch. 👍

commit 5cb4a7f9 (2021-05-26): Install rp2040tools by default // Resolve #8