platformio/platform-raspberrypi #13: Add support for alternative Arduino core earlephilhower/arduino-pico

maxgerhardt, 2021-07-02:
Per https://community.platformio.org/t/request-to-add-platformio-support-for-earle-philhowers-arduino-pico-raspberry-pi-pico-sdk/22285.

As simliar to e.g. STM32 and AVR, it should be supported to switch to a different Arduino core implementation, the most popular of which is https://github.com/earlephilhower/arduino-pico. 

Needs a PlatformIO builder script, a new framwork package and the necessary platform scripts changes here.

sstaub, 2021-09-20:
With the new release (1.9.5) it should possible to integrate in PIO

maxgerhardt, 2021-09-20:
While https://github.com/earlephilhower/arduino-pico/pull/250 has been merged I am blocked by https://github.com/earlephilhower/pico-quick-toolchain/issues/4,  https://github.com/earlephilhower/pico-quick-toolchain/issues/5 and open debugging issues described in https://github.com/earlephilhower/arduino-pico/issues/66#issuecomment-894785078. A WIP version is available at https://github.com/maxgerhardt/pio-pico-core-earlephilhower-test.

Tomwhiting1987, 2021-11-04:
> @sstaub Other than the Pico openocd repo finally merging Picoprobe support into the main branch for unification, not much.
> 
> @Tomwhiting1987 You don't need to go through the GUI to install it, you _have_ to reference the new platform in the `platformio.ini` of the project. Even if you did install through the "Advanced Installation", it wouldn't be picked up if the special `platform = .. ` instruction wasn't there. Best to use the `platformio.ini` from the reference project or the more minimal one shown in the documentation I wrote: https://arduino-pico.readthedocs.io/en/latest/platformio.html

Thanks very much for this Max, much appreciated - seems to be working well now, it's very impressive!    Regarding debugging, is there a way to use pico-debug with the earlephilhower/arduino-pico core in platformio? And if so, how do you do this?

maxgerhardt, 2021-11-04:
@Tomwhiting1987 as of now I even have problems with an extern debug probe, picoprobe, as described in https://github.com/earlephilhower/arduino-pico/issues/66#issuecomment-894785078. There seems to be some (?) issue with the placement of functions in flash compared to RAM. I've not yet touched the picodebug path of uploading or debugging.

Tomwhiting1987, 2021-11-04:
Ok no problem, thanks for the update @maxgerhardt. Is it likely that when this core is fully supported as a platform in platformio, we will be able to use the PIO debug in the GUI?

maxgerhardt, 2021-11-04:
I would only merge it when it's feature-complete with all debugging working, so yes.

sstaub, 2021-11-09:
Why wait for debugging feature? I don't think that all users need it really.

jhmaloney, 2022-01-24:
What's the status of getting this included as an official Platformio framework? Thanks to @maxgerhardt, I've been using it in Platformio for the past several days and it's working great! (I don't need the hardware debugging features.)

I've used both the official, mbed-based framework for the RP2040 and this one by Earle Philhower, and I much prefer this one. For MicroBlocks, the generated code is small and uses much less RAM. Some things, like pin IO, also appear faster. I like the fact that this framework supports LittleFS, although I haven't used that yet, and I'm seeing that Adafruit and others have created libraries around this framework. Finally, I found it easier to dig into the low level code for things like gpio and i2c in the Philhower. (I may have missed something, but I couldn't actually find the source code for the lowest levels in the mbed-based framework.)

In short, I'd love to see this framework supported officially by Platformio.

I've worked a bit with Ivan and the Platformio team in the past, so if it is helpful I can add my voice to those requesting that it be supported officially in whatever way makes sense.

maxgerhardt, 2022-01-24:
We're currently in the process of adding all board JSON files that the Arduino-Pico core supports in https://github.com/earlephilhower/arduino-pico/pull/426, and I've also pinged Earle again on the publishing of the toolchain packages, which are needed.

If I see that getting debugging to work is really too much of a hassle right now, I'll push for a publication without (good / complete) debug support. Gotta prepare for a storm of 'why not working??' issues then though.

episource, 2022-03-13:
I've prepared an [alternative fork](https://github.com/episource/platform-raspberrypi/tree/my-platform-pico) with support for  [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico) : relevant branch is  [episource/platform-raspberrypi#my-platform-pico](https://github.com/episource/platform-raspberrypi/tree/my-platform-pico).

Main differences to maxgerhardt's approach:
 - board definitions and framework build script are not replicated (but read directly from the framework's [tools directory](https://github.com/earlephilhower/arduino-pico/tree/master/tools))
 - builtin PRs #29, #30, #31 for working CMSIS-DAP/SWD transport
 - does not use `toolchain-pico`, but reuses build tools of mbed core  (`toolchain-gccarmnoneeabi`, `tool-openocd-raspberrypi`)
 - automatically disables library `Adafruit TinyUSB Library` if `USE_TINYUSB` is not defined (no need to use `ldf_mode=chain+` or doing this explicitly)
 - core names `pico`, `rp2040` and `earlephilhower` are synonyms
 - no littlefs support though
 - Compiling with `-DUSE_TINYUSB` currently also fails (due to not using `toolchain-pico`)
     * the Adafruit TinyUSB library duplicates definitions from `libpico.a`
     * somehow these duplicates are resolved when using `toolchain-pico`...

jhmaloney, 2022-06-09:
That's great news!

MNS26, 2022-12-22:
any news on this?
i would really like to use the 2nd core

maxgerhardt, 2022-12-22:
Look at the open PRs, it's already there and can be used..

savejeff, 2022-12-22:
> any news on this? i would really like to use the 2nd core
```

[env:BOARD_RP2040_PICO_ALT]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
;platform_packages = framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
board = pico
board_build.core = earlephilhower
board_build.filesystem_size = 0.5m

```