platformio/platform-raspberrypi #14: Add support for Picoprobe

lorenzgeisser, 2021-07-04:
Could you include the Picoprobe as a debugging option in PlatformIO? It would be very useful if we could debug and program the Raspberry Pi Pico (with the RP2040 MCU) projects with an under $5 device.

maxgerhardt, 2021-07-04:
+1 for this. Sadly the RaspberryPi OpenOCD version (https://github.com/raspberrypi/openocd/) seems very fragemented as of now. Looking at the branch `picoprobe` it does have the interface config file (and C code) to support the picoprobe (https://github.com/raspberrypi/openocd/tree/picoprobe/tcl/interface), but this code is missing from the default `rp2040` (maseter) branch. So, compiling one OpenOCD version only gives you one set of features and not some others.

I've opened https://github.com/raspberrypi/openocd/pull/41 as a question (and possible fix) for that. 

Once we have a better OpenOCD version, then this package

https://github.com/platformio/platform-raspberrypi/blob/3c6cc84610e7004eed5980510862a38aa987a7c2/platform.json#L45-L50

can get an update and the needed Python platform.py / main.py code can be added to allow running OpenOCD with the `picoprobe.cfg` interface file, which shouldn't be too hard.

If someone wanted to work with it right-right now one could also use [debug_server](https://docs.platformio.org/en/latest/projectconf/section_env_debug.html#debug-server) and related commands to point to a custom OpenOCD version (compiled from the picoprobe branch) with the appropriate flags, this would very likely work, given that the OpenOCD version in that branch is functional with respect to the picoprobe.

maxgerhardt, 2021-08-08:
I managed to get this working on my own using a custom platform fork and with OpenOCD compiled from the picoprobe branch.

[image]

See **second** environment in my https://github.com/maxgerhardt/pio-pico-core-earlephilhower-test project. Only works on Windows as of now because of a precompiled OpenOCD version with Picoprobe support.

Together with the other improvements in that platform fork (Earle Philhower's Pico core support) I'm planning to PR it back soon.

lorenzgeisser, 2021-08-09:
I took your sample project and was able to compile it successfully (see image 1). But when i test the upload function i get the error "**embedded:startup.tcl:26: Error: Can't find interface/picoprobe.cfg**" (see picture 2). 
Can you help?

[image]

[image]

maxgerhardt, 2021-08-09:
Since the file is [in my repo here](https://github.com/maxgerhardt/pio-openocd-picoprobe/blob/main/share/openocd/scripts/interface/picoprobe.cfg), can you check whether it exists in `C:\Users\<user>\.platformio\packages\tool-openocd-raspberrypi\share\openocd\scripts\interface`?  The output show it's using the OpenOCD version from the Git so it should be there :/

lorenzgeisser, 2021-08-09:
now it works... Thx!
I am looking forward to the picoprobe being officially supported.

flaviut, 2022-03-15:
@maxgerhardt looks like they've combined the two versions in the current rp2040 branch, since there's both `raspberrypi*` and `picoprobe` scripts: https://github.com/raspberrypi/openocd/tree/rp2040/tcl/interface

What would be the next step to getting this implemented? I think it might be to cut a new release onto https://registry.platformio.org/tools/platformio/tool-openocd-raspberrypi, but I'm not sure what needs to be done for that to happen.

I'd be happy to help out if someone would point me in the right direction/do the stuff I don't have permissions for.

maxgerhardt, 2022-06-09:
I've created Picoprobe support in https://github.com/platformio/platform-raspberrypi/pull/36, this is finally working for both the ArduinoCore-mbed core and Arduino-Pico.