platformio/platform-raspberrypi #16: Escape character handling problem

adamalfath, 2021-08-06:
When uploading, the command doesn't handle the spaces on the path.

RP2040 Platform (1.2.0)
tool-rp2040tools (1.0.2)

```
Generating UF2 image
C:\Users\Adam Alfath\.platformio\packages\tool-rp2040tools\elf2uf2 ".pio\build\pico\firmware.elf" ".pio\build\pico\firmware.uf2"
'C:\Users\Adam' is not recognized as an internal or external command,
operable program or batch file.
*** Error 1
Checking size .pio\build\pico\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  18.6% (used 50240 bytes from 270336 bytes)
Flash: [          ]   3.5% (used 72392 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = picotool
Looking for upload port...
Auto-detected: COM56
Forcing reset using 1200bps open/close on port COM56
Uploading .pio\build\pico\firmware.elf
'C:\Users\Adam' is not recognized as an internal or external command,
operable program or batch file.
*** [upload] Error 1
```

valeros, 2021-08-06:
Hi @adamalfath, many thanks for reporting! Should be fixed in the dev branch, please retest with the upstream version of the platform (a [Git client](https://git-scm.com/downloads) is required):
```ini
[env:pico]
platform = https://github.com/platformio/platform-raspberrypi.git
framework = arduino
board = pico
```

closed 2021-08-06 by valeros

maxgerhardt, 2021-08-18:
@valeros actually I have noticed a regressiion with that commit -- very strangely, the uploading won't work then anymore. It always results in 

```
rp2040load 1.0.1 - compiled with go1.15.8
.....................
*** [upload] Error 1
=============================== [FAILED] Took 14.92 seconds ===============================
```

when I go into the platform code and change it back to the full path I get

```
C:\Users\Max\.platformio\packages\tool-rp2040tools\rp2040load -v -D .pio\build\pico_earlephilhower\firmware.elf
rp2040load 1.0.1 - compiled with go1.15.8
.Loading into Flash: [                              ]  0%
Loading into Flash: [=                             ]  6%
Loading into Flash: [===                           ]  12%
Loading into Flash: [=====                         ]  19%
Loading into Flash: [=======                       ]  25%
Loading into Flash: [=========                     ]  31%
Loading into Flash: [===========                   ]  38%
Loading into Flash: [=============                 ]  44%
Loading into Flash: [===============               ]  50%
Loading into Flash: [=================             ]  57%
Loading into Flash: [==================            ]  63%
Loading into Flash: [=====================         ]  70%
Loading into Flash: [======================        ]  76%
Loading into Flash: [========================      ]  82%
Loading into Flash: [==========================    ]  89%
Loading into Flash: [============================  ]  95%
Loading into Flash: [==============================]  100%

=============================== [SUCCESS] Took 4.51 seconds ===============================
```

Very strange.. Maybe the program doesn't like being called via the PATH and there are some current-working-directory (cwd) issues.

valeros, 2021-08-19:
@maxgerhardt Indeed a strange problem, I reversed the changes and wrapped the path to upload tools, so now it should work as expected. Thanks for pointing it out.

commit 7b48a72f (2021-08-06): Simplify paths to platform tools to handle whitespaces

Resolve #16