platformio/platform-raspberrypi #17: Update for ArduinoCore-mbed

sstaub, 2021-08-14:
Can you please update the ArduinoCore-mbed to v2.4.1

valeros, 2021-08-17:
Updated, please retest with the [upstream version](https://docs.platformio.org/en/latest/platforms/raspberrypi.html#upstream) of the platform.

closed 2021-08-17 by valeros

sstaub, 2021-08-17:
Can compile. If I want to use the upstream version I have to update manually the platform. Is this intended?
When I change to upstream version, I think the new files should automatically updated, when changing the .ini and first compile or build.

valeros, 2021-08-17:
> If I want to use the upstream version I have to update manually the platform. Is this intended?

No, it's not. PlatformIO should handle your platform specification in the `platformio.ini` file automatically.

sstaub, 2021-08-17:
Maybe a bug?
I changed the platform to
platform = https://github.com/platformio/platform-raspberrypi.git
but when I build the old version was used. I'm using autosave for the files.

valeros, 2021-08-17:
Please attach here your `platformio.ini`.

sstaub, 2021-08-17:
```
[env:pico]
#platform = raspberrypi
platform = https://github.com/platformio/platform-raspberrypi.git
board = pico
framework = arduino
```

valeros, 2021-08-17:
Probably you have multiple `raspberrypi` platforms installed in your system. Are there any suspicious folders with the `raspberrypi@hash` suffix in `~/.platformio/platforms`? If so, you can either delete them or run `pio platform update raspberrypi` to pull the latest changes.

sstaub, 2021-08-17:
Ok, it seems I was too fast. Maybe I started building while a background task was running. Cleaned the piofolder and tried again, now it works. Thank You

commit 39b2ae97 (2021-08-17): Update Arduino-mbed package to v2.4.1

Resolve #17