platformio/platform-raspberrypi #18: Add support for FreeRTOS

maxregan, 2021-10-06:
As of https://github.com/FreeRTOS/FreeRTOS-Kernel/pull/341, FreeRTOS now has support for the RP2040. It would be great if platformio enabled FreeRTOS as a framework for RP2040.

KenwoodFox, 2022-07-27:
Been a while, did this get merged? PIO Still says FreeRTOS is unsupported.

sstaub, 2022-07-27:
Only the Earl E. Philhower core supports FreeRTOS. The mbed core is an RTOS.

KenwoodFox, 2022-07-27:
I did not realize mbed core was an rtos. Thats very interesting i should explore that, but, the rp2040 should still support FreeRTOS right?

Im not an expert with scons but i dug around a little, i can kinda see where the new builder goes for adding freertos ontop of bare? Not sure exactly how to go about forking and testing some changes yet.

sstaub, 2022-07-27:
FreeRtOS will not support by the original Arduino core which is based on mbd.
Have a look to https://github.com/earlephilhower/arduino-pico
The core is not yet official supported directly from PIO but will merged soon, I hope.
See also the docs https://arduino-pico.readthedocs.io/en/latest/

KenwoodFox, 2022-07-27:
Ah ok i understand now, thanks so much! 

Honestly this looks super fun on its own even without an RTOS~ https://arduino-pico.readthedocs.io/en/latest/multicore.html

Hopefully we can see some offical support sometime, for now I think this clears it up for some people exploring options here.

savejeff, 2022-07-29:
i work with the earlephilhower pico core a lot with multicore. it works very well and stable with some recent updates.
i heared this core supports Rtos threads but i to this point only used both cores (which is very vell supported on this core)

I recommend the arduino-pico core to everybody. its a single line in platformio.ini and you are using a much better constructed arduino core for the rp2040. the maintainer is also very fast and helpful with issues

KenwoodFox, 2022-07-30:
> i work with the earlephilhower pico core a lot with multicore. it works very well and stable with some recent updates. i heared this core supports Rtos threads but i to this point only used both [...]

Im actually having a little trouble getting earlephilhower's pico core running at the moment, dosent seem to like being added directly to platformio? Whats the line maybe i have it wrong, i'd love to get this up and running and learn more, very excited to try out the second core (and maybe some rtos "threads" if i can figure it out)

Good task management would make me super happy

VikasPoddar, 2022-10-19:
Is platformio going to release support for `framework = freertos` in platfromio ???