platformio/platform-raspberrypi #24: printf not working with RP2040

fabltd, 2022-01-02:
Hi 

I have some example RP2040 code that works in the Ardunio IDE without fail. When ported to Platform IO the code fails on the following:

Serial.printf("timers_init failed - %d\n", err_code);

```
src/main.cpp: In function 'void loop()':
src/main.cpp:22:10: error: 'class arduino::UART' has no member named 'printf'; did you mean 'print'?
   22 |   Serial.printf("timers_init failed - %d\n", err_code);

```
This works in the ardinio IDE.

Thanks

fabltd, 2022-01-02:
Try the following in Ardunio IDE - Works ! - Does not work in Platform IO!!

```
void setup() {
  time_t timeout = millis();
  Serial.begin(115200);
  while (!Serial)
  {
    if ((millis() - timeout) < 5000)
    {
      delay(100);
    }
    else
    {
      break;
    }
  }

}

void loop() {
  int err_code = 1;
  Serial.printf("timers_init failed - %d\n", err_code);

}
```

fabltd, 2022-01-02:
Not sure if it helps but is the complie debug:

```
compiling .pio/build/rak11300/FrameworkArduino/main.cpp.o
In file included from /Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/platform_defs.h:12,
                 from /Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/pico_platform/include/pico/platform.h:12,
                 from /Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/common/pico_base/include/pico.h:19,
                 from /Users/jeremy/.platformio/packages/framework-arduino-mbed/variants/RASPBERRY_PI_PICO/double_tap_usb_boot.cpp:2:
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'uint32_t time_us_32()':
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:66:12: note: in expansion of macro 'timer_hw'
   66 |     return timer_hw->timerawl;
      |            ^~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h: In function 'bool time_reached(absolute_time_t)':
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:110:19: note: in expansion of macro 'timer_hw'
  110 |     uint32_t hi = timer_hw->timerawh;
      |                   ^~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_regs/include/hardware/regs/addressmap.h:56:20: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
   56 | #define TIMER_BASE 0x40054000
      |                    ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2040/hardware_structs/include/hardware/structs/timer.h:33:38: note: in expansion of macro 'TIMER_BASE'
   33 | #define timer_hw ((timer_hw_t *const)TIMER_BASE)
      |                                      ^~~~~~~~~~
/Users/jeremy/.platformio/packages/framework-arduino-mbed/cores/arduino/mbed/targets/TARGET_RASPBERRYPI/TARGET_RP2040/pico-sdk/rp2_common/hardware_timer/include/hardware/timer.h:111:33: note: in expansion of macro 'timer_hw'
  111 |     return (hi >= hi_target && (timer_hw->timerawl >= (uint32_t) target || hi != hi_target));
      |                                 ^~~~~~~~
Compiling .pio/build/rak11300/FrameworkArduino/mbed/platform/cxxsupport/mstd_mutex.cpp.o
Compiling .pio/build/rak11300/FrameworkArduino/pinToIndex.cpp.o
src/main.cpp: In function 'void loop()':
src/main.cpp:22:10: error: 'class arduino::UART' has no member named 'printf'; did you mean 'print'?
   22 |   Serial.printf("timers_init failed - %d\n", err_code);
      |          ^~~~~~
      |          print
Compiling .pio/build/rak11300/FrameworkArduino/wiring.cpp.o
*** [.pio/build/rak11300/src/main.cpp.o] Error 1
```

Matthew-McRaven, 2022-04-19:
Also seeing this on a Arduino Due

ark-, 2022-05-10:
Getting this on a Pi Pico also. It seems that printf isn't included in the RP2040 version of the Arduino framework.

LeoDJ, 2022-05-12:
You can use the `REDIRECT_STDOUT_TO()` macro to enable the "real" printf.

Disclaimer: On USB Serial at least, the printed characters of the current line only get sent once a `\n` characters appears.

Minimal working example:
```C
#include <Arduino.h>

REDIRECT_STDOUT_TO(Serial); 

void setup() {
    delay(5000);
    printf("Hello World!\n");
}

void loop() {
    printf("%ld\n", millis());
    delay(1000);
}
```

platformio.ini:
```ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
```

maxgerhardt, 2022-06-09:
`Serial.printf()` is **not** available in the ArduinoCore-mbed, but in Arduino-Pico. Support for that core will be added in #13.