platformio/platform-raspberrypi #25: Add support for Seeed Xiao RP2040 board

colingimenez, 2022-01-03:
Would it be possible to add support for the Seeed Xiao RP2040 board? [Here's a link to it](https://www.seeedstudio.com/XIAO-RP2040-v1-0-p-5026.html).

I tried adapting the Arduino package that Seeed provides but I just don't understand it well enough to know what exactly I need to do. [This](https://files.seeedstudio.com/arduino/core/rp2040/Seeed_RP2040_boards.1.9.3.tar.bz2) is the one I was trying to work with. I would be willing to give it another go myself if someone could take a look at it and give me a more detailed explanation than what the PlatformIO documentation provides.

Thanks!

maxgerhardt, 2022-06-10:
Support through the official Arduino-Pico core's Seeed Xiao RP2040 variant will be added in #36.

colingimenez, 2022-06-10:
Fantastic. Thanks Max!