platformio/platform-raspberrypi #26: I have a built error when I include "Keyboard.h"

Fu-Xiaotian, 2022-01-06:
`C:\Users\76504\.platformio\packages\framework-arduinopico\libraries\Keyboard\src\keyboard.cpp: In member function 'void Keyboard_::sendReport(KeyReport*)':
C:\Users\76504\.platformio\packages\framework-arduinopico\libraries\Keyboard\src\keyboard.cpp:53:5: error: 'tud_task' was not declared in this scope
   53 |     tud_task();
      |     ^~~~~~~~
*** [.pio\build\pico\libb00\Keyboard\Keyboard.cpp.o] Error 1`
This is my PIO.ini
`[env:pico]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = arduino
board_build.core = earlephilhower
board_build.f_cpu = 133000000L
board_build.filesystem_size = 0m
platform_packages = 
	maxgerhardt/framework-arduinopico@https://github.com/earlephilhower/arduino-pico.git
	maxgerhardt/toolchain-pico@http://github.com/earlephilhower/pico-quick-toolchain/releases/download/1.3.1-a/x86_64-w64-mingw32.arm-none-eabi-7855b0c.210706.zip`

Fu-Xiaotian, 2022-01-06:
I just fix it by delete the TinyUSB libraries from .platformio\packages\framework-arduinopico\libraries
If tinyusb library is not used, you should delete the libraries to avoid PIO link wrong source.
I hope there will be a solution to this problem in the future.

maxgerhardt, 2022-06-09:
This was very likely fixed by https://github.com/earlephilhower/arduino-pico/pull/513.

closed 2022-06-18 by Fu-Xiaotian