platformio/platform-raspberrypi #28: Temporary solution for FW uploading in windows

Gusev-Roman, 2022-02-20:
== platformio.ini
[env:pico]
platform = raspberrypi
board = pico
framework = arduino
upload_protocol = mbed
upload_port = E:

== platforms/raspberrypi/builder/main.py
`if upload_protocol == "mbed":
    upload_actions = [
        env.VerboseAction(env.AutodetectUploadPort, "Looking for upload disk..."),
        env.VerboseAction(env.UploadToDisk, "Uploading $SOURCE")
    ]`

must be replaced to

`if upload_protocol == "mbed":
    target_firm = join("$BUILD_DIR", "${PROGNAME}.uf2")
    print('Copying file:')
    env.Execute("copy " + upload_source + " $UPLOAD_PORT")
    upload_actions = [
        env.VerboseAction(env.AutodetectUploadPort, "Looking for upload disk..."),
        #env.VerboseAction(env.UploadToDisk, "Uploading $SOURCE")
    ]`

now uploading works!..

oomek, 2022-03-12:
Thanks, but now I gave to unplug and hold the button. Is there any way to join both actions: forcing the board into bootsel from picotool and copying the file as you described?

maxgerhardt, 2022-06-09:
`upload_protocol = mbed` upload is fixed by #36.

>Is there any way to join both actions: forcing the board into bootsel from picotool and copying the file as you described?

Once a firmware from the Arduino core has been uploaded, it accepts a 1200bps reset-to-bootloader through the USB Serial interface it creates. Just delete any `upload_protocol = ..` line from the `platformio.ini` after, it's the default protocol.