platformio/platform-raspberrypi #34: Under PlatformIO, debug rp2040 with "rp2040 dap" is ok, but download shows error.

kylongmu, 2022-05-10:
Make a "rp2040 dap" with [this](https://github.com/ataradov/free-dap/blob/master/bin/free_dap_rp2040.uf2) , and use it to debug another rp2040 is very cool, debug is OK, but download shows error.
This is my platform.ini:

```
[env:upstream_develop]
platform = https://github.sunflyer.cn/platformio/platform-raspberrypi.git
framework = arduino
board = pico

board_build.f_cpu = 120000000L
debug_tool = cmsis-dap
upload_protocol = cmsis-dap
debug_speed = 1000000
```
This is the upload log:

```
> Executing task in folder 220428-141324-arduino-blink: C:\Users\THINKPAD\.platformio\penv\Scripts\platformio.exe run --target upload <

Processing upstream_develop (platform: https://github.sunflyer.cn/platformio/platform-raspberrypi.git; framework: arduino; board: pico)
----------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/raspberrypi/pico.html
PLATFORM: Raspberry Pi RP2040 (1.6.0+sha.e08da69) > Raspberry Pi Pico
HARDWARE: RP2040 120MHz, 264KB RAM, 2MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, jlink, raspberrypi-swd)
PACKAGES:
 - framework-arduino-mbed 3.0.1
 - tool-openocd-raspberrypi 2.1100.0 (11.0)
 - tool-rp2040tools 1.0.2
 - toolchain-gccarmnoneeabi 1.90201.191206 (9.2.1)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 37 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <SPI>
Building in release mode
Checking size .pio\build\upstream_develop\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.0% (used 45844 bytes from 270336 bytes)
Flash: [          ]   0.2% (used 4322 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\upstream_develop\firmware.elf
Open On-Chip Debugger 0.11.0+dev-gec0240e6a-dirty (2021-03-09-09:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 1000000 kHz

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10013fa0
** Programming Finished **
** Verify Started **
Warn : target was in unknown state when halt was requested
Error: timed out while waiting for target halted
Error: error executing cortex_m crc algorithm
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530
*** [upload] Error 1
========================================================= [FAILED] Took 5.58 seconds ====================================
```

maxgerhardt, 2022-06-09:
`Error: checksum mismatch - attempting binary compare` I've experienced this error too when using a too high programming speed. 
```
debug_speed = 1000000
```
this value is in khz. 1 million killohertz is a gigahertz. Please set `debug_sped = 1000` and retry debugging, or try out the PR #36, it also respects the speed when using a regular upload.

kylongmu, 2022-06-10:
> `Error: checksum mismatch - attempting binary compare` I've experienced this error too when using a too high programming speed.
>
> ```
> debug_speed = 1000000
> ```
>
> this value is in khz. 1 million [...]

But the download is not work also, this is the output with  `debug_sped = 1000` 
```
RAM:   [==        ]  17.3% (used 46672 bytes from 270336 bytes)
Flash: [          ]   0.2% (used 4346 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\upstream_develop\firmware.elf
Open On-Chip Debugger 0.11.0+dev-gec0240e6a-dirty (2021-03-09-09:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 1000 kHz

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10016ad8
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x1000c9ac psp: 0x20008838
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530
*** [upload] Error 1
======================================================== [FAILED] Took 23.32 seconds ========================================================
```

maxgerhardt, 2022-06-10:
Can you try to go even lower, like 100kHz?

maxgerhardt, 2022-06-10:
Does this also occur when using the Picoprobe firmware (https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#debugging-using-another-raspberry-pi-pico) and configuration? (`upload_protocol = picoprobe` and wire them up like seen in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf page 61)

maxgerhardt, 2022-06-10:
Instead of using `platform = https://github.sunflyer.cn/platformio/platform-raspberrypi.git` you can try using `platform = https://github.com/maxgerhardt/platform-raspberrypi.git`, so that you have a more up-to-date openocd version and picoscope support. Maybe that fixes it already.

kylongmu, 2022-06-10:
> 

With 100k download:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.3% (used 46672 bytes from 270336 bytes)
Flash: [          ]   0.2% (used 4346 bytes from 2097152 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picotool, raspberrypi-swd
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\upstream_develop\firmware.elf
Open On-Chip Debugger 0.11.0+dev-gec0240e6a-dirty (2021-03-09-09:44)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 100 kHz

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10016ad8
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x81000000 pc: 0x1000c9ac psp: 0x20008838
Error: checksum mismatch - attempting binary compare
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530
*** [upload] Error 1
======================================================== [FAILED] Took 38.31 seconds ========================================================
```
And can't enter debug mode with 100k

kylongmu, 2022-06-10:
> Instead of using `platform = https://github.sunflyer.cn/platformio/platform-raspberrypi.git` you can try using `platform = https://github.com/maxgerhardt/platform-raspberrypi.git`, so that you have a more up-to-date openocd version and picoscope support. Maybe that fixes it already.

This is OK:
```
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.3% (used 46672 bytes from 270336 bytes)
Flash: [          ]   0.2% (used 4346 bytes from 2093056 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, jlink, picoprobe, picotool, raspberrypi-swd
CURRENT: upload_protocol = cmsis-dap
Uploading .pio\build\upstream_develop\firmware.elf
Open On-Chip Debugger 0.11.0-g610f137 (2022-02-11-13:57)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

adapter speed: 1000 kHz

target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000184 msp: 0x20041f00
** Verified OK **
** Resetting Target **
shutdown command invoked
======================================================== [SUCCESS] Took 33.74 seconds ========================================================
```
Thanks a lot!

maxgerhardt, 2022-06-10:
Then the issue will be fixed once https://github.com/platformio/platform-raspberrypi/pull/36 is merged.