platformio/platform-raspberrypi #37: When start the second core of rp2040,  MbedOS Fault and halt.

kylongmu, 2022-06-14:
This is my test code for multicore, with PIO arduino framework:

```
#include "Arduino.h"
#include "pico/multicore.h"
void setup() {
  multicore_reset_core1();
  multicore_launch_core1(core1_entry);
}
void loop() {
}
void core1_entry() {
  pinMode(LED_BUILTIN, OUTPUT);
  while(1) {
    digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
    delay(200);               // wait for a second
    digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
    delay(200);               // wait for a second
  }
}

```
But the code halt at the start, and serial output these:

```
InitPioProgram!

++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0   : FFFFFFDC
R1   : 2000B970
R2   : 2000B92C
R3   : 20000448
R4   : 000000C8
R5   : 00000019
R6   : 00000179
R7   : 1000355D
R8   : FFFFFFFF
R9   : FFFFFFFF
R10  : FFFFFFFF
R11  : FFFFFFFF
R12  : FFFFFFFF
SP   : 20040FB0
LR   : FFFFFFF9
PC   : 10000334
xPSR : A100000B
PSP  : FFFFFFFC
MSP  : 20040F90
CPUID: 410CC601
Mode : Handler
Priv : Privileged
Stack: MSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x10000334
Error Value: 0x2000B7A4
Current Thread: rtx_idle <handler> Id: 0x2000B92C Entry: 0x100055FD StackSize: 0x200 StackMem: 0x200088C0 SP: 0x20040FB0 
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&osver=61501&core=0x410CC601&comp=2&ver=110300&tgt=RASPBERRY_PI...

-- MbedOS Error Info --
```

Is the muticore not be supported?

kylongmu, 2022-06-14:
Try another multicore example from [here](https://github.com/earlephilhower/arduino-pico/blob/master/libraries/rp2040/examples/Multicore/Multicore.ino).
But the core1 without any output.
Reference [Doc](https://arduino-pico.readthedocs.io/en/latest/multicore.html).

sstaub, 2022-06-14:
You must use the Earl E. Philhower core. mbed does not support the official sdk.