platformio/platform-raspberrypi #45: Request/Question: Baremetal core without Arduino

jonathanlundstrom, 2023-03-28:
Hi there,

How hard would it be to adapt this platform to only bundle the Pico SDK, debug tools and other nice things from this project? The reason for my question being that we want to avoid the bloated and inefficient Arduino framework all together and only work with C and the Pico SDK but retain the ability to build and debug our code. We've previously relied on Wizio-Pico but that project has since been scrapped.

Thank you!

Regards,
Jonathan

jonathanlundstrom, 2023-03-28:
Thank you. I'm aware of this one as well – but I dread Arduino and do not want it anywhere near my hardware. The earlephilhower core has tons of crippling Wi-Fi and socket issues that are just not possible to work around. I want to run a pure C core with only the Pico SDK and nothing else.

savejeff, 2023-03-28:
then you will have to use the sdk directly. just use bare Visual studio code with c++ extension for intelisense. 
platformio is very rarely offering not arduino sdks and i understand that desission. arduino ist way more unified and easier maintainable. maybe for the pico SDK there will be an exception like for the esp-idf but not in the near future.

about the wifi socket issues, could you elaborate. im planing to work on wifi stuff on rp2040 in the future, would be good to know what problems are known.

also about power consumption, that rp2040 is just not that good at low power mode. you will not be able to go below 1mA

RudolphRiedel, 2023-06-03:
+1 for offering Baremetall as well.
I also used to use Wizio-Pico and unfortunately found that to be defunct now.
And no, I do not have any appreciation for the decision to not allow any other framework than Arduino and even going so far to block pull-requests like with the Atmel-Sam platform.
How is offering a CMSIS or native SDK or whatever you want to call it framework difficult to maintain when all that is necessary for it is already included in the Arduino framework?

Ironman-007, 2024-10-08:
[env:stable]
platform = https://github.com/maxgerhardt/platform-raspberrypi.git
board = pico
framework = picosdk
; framework = arduino
board_build.core = earlephilhower

debug_tool = jlink
upload_protocol = jlink

this works for me, I can use the sdk to develop the firmware for 2040, but not sure if this has limited functions and also not sure if this support the PIO.