platformio/platform-raspberrypi #55: build fails, trips over commented-out include

RudolphRiedel, 2023-09-30:
I just ran into the issue that the Arduino example for my library does not build anymore for the Raspberry Pi Pico using this Arduino core.
And after some digging I have a minimal example that does fail to build.

[Arduino_RP2040_build_fail.zip](https://github.com/platformio/platform-raspberrypi/files/12774574/Arduino_RP2040_build_fail.zip)

This is the code:
#include <Arduino.h>

#if 0
#include "Bsp.h"
#endif

void setup()
{
}

void loop()
{
}

And the result is that somehow a "BSP.c" for the Arduino Portenta gets compiled which failed
as stm32h7xx_hal.h can not be included.

Is this some dependency scan gone wrong?

The code I actually have an issue with is more complicated, I have this include file:
https://github.com/RudolphRiedel/FT800-FT813/blob/5.x/EVE_target/EVE_target_Tricore_Tasking.h

It is not touched by anything when building for Arduino RP2040 and it starts with this:

---
#ifndef EVE_TARGET_TRICORE_H
#define EVE_TARGET_TRICORE_H

#if !defined (ARDUINO)
#if defined(__TASKING__)

#include <stdint.h>

#include "IfxPort.h"
#include "Bsp.h"
---

So even if it would be included somehow, the build that does not work is for Arduino and using GCC, not the Tasking compiler.
If I delete this file the project builds just fine - so it really is not included - but as far as I know I can not tell PlatformIO to exclude files from a library for specific builds.

RudolphRiedel, 2023-09-30:
I just tried to build the example above with other releases of this Platform including the current repository.
The versions below 1.5.0 can not be installed anymore.
And every version from 1.6.0. to current has the same issue.

valeros, 2023-09-30:
Hi @RudolphRiedel, by default Library Dependency Finder does not evaluate C/C++ conditional syntax. In your case you need to add the following line `lib_ldf_mode = chain+` to your `platformio.ini` to enable the preprocessor. More info here https://docs.platformio.org/en/latest/librarymanager/ldf.html#ldf-mode

closed 2023-09-30 by valeros

RudolphRiedel, 2023-09-30:
Is there a way to change this outside of the platformio.ini?
How am I supposed to communicate this to the users of my library?

And why is evaluating the C/C++ preprocessor conditional syntax not the default mode, why is it preferred to compile unrelated code into a project?
This must be happening all the time, I really do wonder why this does not fail more often.

valeros, 2023-10-01:
> Is there a way to change this outside of the platformio.ini?
> How am I supposed to communicate this to the users of my library?

You need to create a special library manifest file called `library.json` and specify a proper LDF mode in the `build` section of that file. More info about `library.json` can be found here https://docs.platformio.org/en/latest/librarymanager/index.html , an example of such a file is here https://github.com/jgromes/RadioLib/blob/master/library.json

RudolphRiedel, 2023-10-01:
Thank you, this worked just fine .

One last question though, do the settings in the library.json only apply to the code of the library?

I looks like it does, even setting "lib_ldf_mode = chain" in the platformio.ini does not prevent my project from building now, but the documentation does not specify.

valeros, 2023-10-02:
> One last question though, do the settings in the library.json only apply to the code of the library?

That's right, only the source code of the library will be analyzed in the specified LDF mode.