platformio/platform-raspberrypi #60: I need help in downloading PICO SDK to Platform IO so I can use SDK specific functions as I cannot already.

tybasuleman, 2023-11-15:
What kind of issue is this?

- [x] **Question**.
      This issue tracker is not the place for questions. If you want to ask how to do something,
      or to understand why something isn't working the way you expect it to, 
      use [Community Forums](https://community.platformio.org) or [Premium Support](https://platformio.org/support)

- [x] **PlatformIO IDE**.
      All issues related to PlatformIO IDE should be reported to the
      [PlatformIO IDE for VSCode](https://github.com/platformio/platformio-vscode-ide/issues) repository

- [x] **Development Platform or Board**.
      All issues (building, uploading, adding new boards, etc.) related to PlatformIO development platforms
      should be reported to appropriate repository related to your hardware
      https://github.com/topics/platformio-platform

- [x] **Feature Request**.
      Start by telling us what problem you’re trying to solve. Often a solution
      already exists! Don’t send pull requests to implement new features without first getting our
      support. Sometimes we leave features out on purpose to keep the project small.

- [x] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

*You can erase any parts of this template not applicable to your Issue.*

------------------------------------------------------------------

### Configuration

**Operating system**:

**PlatformIO Version** (`platformio --version`):

### Description of problem

#### Steps to Reproduce

1.
2.
3.

### Actual Results

### Expected Results

### If problems with PlatformIO Build System:

**The content of `platformio.ini`:**
```ini
Insert here...
```

**Source file to reproduce issue:**
```cpp
Insert here...
```

### Additional info

valeros, 2023-11-15:
Currently, there is no support for Pico SDK, closing as duplicate #45.

closed 2023-11-15 by valeros

tybasuleman, 2023-11-15:
> Currently, there is no support for Pico SDK, closing as duplicate #45.

Okay, but the documentation of platform says #[here](https://arduino-pico.readthedocs.io/en/latest/sdk.html#) that full functionalities of Pico SDK are being supported.

valeros, 2023-11-15:
That's the documentation for a different Arduino core that is not currently supported, the `raspberrypi` dev-platform uses the official one from the Arduino team https://github.com/arduino/ArduinoCore-mbed

tybasuleman, 2023-11-15:
> That's documentation for a different Arduino core that is not currently supported, the `raspberrypi` dev-platform uses the official one from the Arduino team https://github.com/arduino/ArduinoCore-mbed

Could you help me to get the arduino core that supports PICO SDK in PlatformIO?

valeros, 2023-11-16:
There is no official support, but you can take a look at this implementation https://github.com/platformio/platform-raspberrypi/pull/36

tybasuleman, 2023-11-16:
Alright, Thank You