platformio/platform-raspberrypi #67: [REQUEST] Raspberry Pi RP2350 and Raspberry Pi Pico 2 support?

Hedda, 2024-08-13:
Please consider looking into adding platform support for the new RP2350 MCU and the Raspberry Pi Pico 2 development board.

RP2350 is the successor and next-generation of RPi RP2040 so guess would need to update the RP2040 platform for it or add another platform?

RP2350 will be generally available in volume before the end of 2024. To register your interest, and to participate in the Raspberry Pi samples program, head over to the [product page](https://www.raspberrypi.com/products/rp2350/).

FYI, a "W" version of the Pico 2 with wireless support expected 'by the end of the year' according the people in the know 😛

Raspberry Pi team writes that the new [RP2350](https://www.raspberrypi.com/products/rp2350/) is vastly more sophisticated design than RP2040, however I understand it is supposedly backwards-compatible, though this new model now also offers faster resources, more IO/PIO,  support for Arm TrustZone (security architecture), new HSTX peripheral for high-speed data transmission, and support for external QSPI PSRAM.

But I believe that the huge difference is otherwise the fact that the new Raspberry Pi RP2350 MCU features two-modes with dual-core RISC-V or dual-core ARM Cortex-M33 on the chip, meaning that you can either choose to run it in either RISC-V mode or in ARM Cortex-M3 mode, (but not more than two cores active at the same time, so **_"They're selectable at boot time: Each port into the bus fabric can be connected either to an M33 or a Hazard3 via a mux. You can even, if you're feeling obtuse, run with one of each."_** Arm/RISC-V switching is explained in the official datasheet:

* https://www.raspberrypi.com/news/raspberry-pi-pico-2-our-new-5-microcontroller-board-on-sale-now/
* https://www.raspberrypi.com/products/raspberry-pi-pico-2/
* https://www.raspberrypi.com/documentation/microcontrollers/pico-series.html#pinout-and-design-files
* https://www.raspberrypi.com/products/rp2350/
* https://datasheets.raspberrypi.com/rp2350/rp2350-datasheet.pdf
* https://github.com/raspberrypi/pico-sdk-rp2350
* https://github.com/raspberrypi/pico-examples-rp2350
* https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf
* https://github.com/raspberrypi/pico-vscode

Many RP2350-based development boards and products are apparently already in the pipeline:

https://www.raspberrypi.com/for-industry/powered-by/product-catalogue/?category=RP2350

If official Pico2 is hard to find then be ware that SparkFun Pro Micro RP2350 (and DEV-24870)  and Adafruit Feather RP2350 (Product ID: 6000) are supposedly among the first of third-parties  scheduled to become available to the first third-party boards available with RP2350:

* https://www.sparkfun.com/products/24870

This article from CNX software offers a little more insight into the RP2350 as compared to the RP2040:

* https://www.cnx-software.com/2024/08/08/raspberry-pi-pico-2-raspberry-pi-rp2350-dual-core-risc-v-or-arm-cortex-m33-microcontroller/

"**_The RP2350 embeds both an open-source Hazard3 RISC-V dual-core CPU and a dual-core Cortex-M33, but only one cluster can be used at a given time. Apart from the faster MCU cores and higher SRAM capacity, the RP2350 is about the same as the RP2040, albeit it also adds one extra PIO block bringing the total to three. One important new feature is built-in security when using Arm Cortex-M33 cores with Trustzone and other security features._**"

**_As I understand it, the RP2350A package offers also the same pinout as the RP2040 microcontroller, but the company now also adding a larger RP2350B package with additional GPIOs and analog inputs._**

_**Raspberry Pi used the same method as for the RP2040 to derive the RP2350 name. RP stands for “Raspberry Pi”, “2” is the number of cores, “3” refers to the MCU core used (e.g. Cortex-M33), and the last two numbers “4” and “0” use floor(log2(x/16k)) formula to calculate a number representing the SRAM and non-volatile storage capacity inside the chip.**_

* https://www.raspberrypi.com/documentation/microcontrollers/silicon.html#architecture-switching

[image]

[image]

[image]

[image]

[image]

PS: Off-topic; Raspberry Pi is btw offering a $10,000 bounty for first confirmed break of their new signed boot process on RP2350.:

https://github.com/raspberrypi/rp2350_hacking_challenge

"_**To get RP2350 hardware into the hands of the engineers most likely to find these flaws, we’ve partnered with the [DEF CON hacking convention](https://defcon.org/), which starts today in Las Vegas. This year’s badge is powered by RP2350, and makes a great platform for experimenting with our security architecture.**_"

PPS: Ustream forum discussions from before when they leaked the news about this "Forthcoming RP235x" microcontroller:

* https://forums.raspberrypi.com/viewtopic.php?t=371165

Hedda, 2024-08-13:
For reference, here is a related feature request to the [arduino-pico](https://github.com/earlephilhower/arduino-pico) project:

* https://github.com/earlephilhower/arduino-pico/issues/2325

More information in this discussion:

* https://github.com/earlephilhower/arduino-pico/discussions/2226

FYI, [PontusO](https://github.com/PontusO) has a forked arduino-pico to a new arduino-pico-rp2350 repo where he got RP2340 working with an older version of the pico-sdk:

https://github.com/PontusO/arduino-pico-rp2350

> _An arduino-pico port is on it's way (https://github.com/PontusO/arduino-pico-rp2350). Unfortunatly there were a ton of PR's on the pico-sdk the past few days so it is not ready for prime time yet. **It works fairly well in its current state though but this is using a 2 month old version of the pico-sdk**._

_Originally posted by [PontusO](https://github.com/PontusO) in https://github.com/earlephilhower/arduino-pico/discussions/2226#discussioncomment-10278437_

We do not know yet if he will leave that new arduino-pico-rp2350 repo as a a separate fork dedicated only to the RP2350 or if he or others will submit pull requests with changes from it to earlephilhower’s arduino-pico repo and use that as upstream for all picos, including both RP2040 and RP2350 in the same repository.

https://github.com/PontusO) in https://github.com/earlephilhower/arduino-pico/discussions/2226

Hedda, 2024-09-06:
FYI, this fork by @maxgerhardt has now added RP2350 building + debugging support with latest Arduino-Pico core as default:

* https://github.com/maxgerhardt/platform-raspberrypi/issues/69

Also note that RP2350 is now supported by arduino-pico (alternative Raspberry Pi Pico Arduino core library by @earlephilhower):

* https://github.com/earlephilhower/arduino-pico/issues/2325

znmeb, 2024-09-23:
> From what I just read, PlatformIO will probably never add add RP2350 support or accept community contributions for this?

...

> FYI, instead use fork by @maxgerhardt which added RP2350 building + debugging with latest Arduino-Pico core as default:

Is that the final answer? I have a project using that repo now but it's a maintenance nightmare to require a non-released version of a developer tool.

https://github.com/AlgoCompSynth/cforth/tree/master/RP2350-testing

Hedda, 2024-09-23:
> FYI, instead use fork by @maxgerhardt which added RP2350 building + debugging with latest Arduino-Pico core as default:
>
>> Is that the final answer? I have a project using that repo now but it's a maintenance nightmare to require a non-released version of a developer tool.

I do not work PlatformIO or Raspberry Pi so I can not give you any answers. I am only an entusiast and hobbyist in regards to this, and this issue is just a feature request.

Not sure if you will get a official answer for PlatformIO or not, however, PlatformIO's CEO have more or less made it clear that they will not allow their upstream platform-raspberrypi to keep being developed in their repository/repositories as long as Raspberry Pi Foundation or Raspberry Pi Ltd is not officially paying PlatformIO to maintain it there, and infer that the decision to discontinue support for upstream PlatformIO platforms are entirly up to the companies paying PlatformIO to maintain each platform.

The alternative options for now is to either use a fork or use a Raspberry Pi Pico Arduino core without PlatformIO. New forks of PlatformIO's platform libraries have been created due to the lack of ongoing development in PlatformIO upstream platform library. Anyone in the community can use and contribute to those, see example:

* https://github.com/maxgerhardt/platform-raspberrypi

FYI, it looks like we can see practically the same situation between PlatformIO and Espressif (platform-espressif32 for ESP32) which this has gone on for longer:

* https://github.com/platformio/platform-espressif32/issues/1225

PlatformIO is also no longer taking community contributions to improve its platform-raspberrypi platform library in their upstream repository/repositories, which means that upstream development of PlatformIO's platform library for Raspberry Pi Pico(s) have stalled or put on hold for now. Because of this users feel forced to move to forks or alternative Arduino-Pico core without PlatformIO in order to use the new RP2350 MCU/SoC. 

* https://github.com/maxgerhardt/platform-raspberrypi/issues/73

* https://github.com/earlephilhower/arduino-pico
  * https://arduino-pico.readthedocs.io/en/latest/

 For additional information, please refer to these GitHub links where the discussions there are self-explanatory, allowing you to draw your own conclusions:

* https://github.com/espressif/arduino-esp32/discussions/10039
* https://github.com/platformio/platform-espressif32/issues/1225
* https://github.com/espressif/arduino-esp32/pull/8606

PS: I understand PlatformIO is a company that need income but I do not see how this practice will benefit them in the long run.

znmeb, 2024-09-23:
Ah, OK. I only have one project using PlatformIO: https://github.com/AlgoCompSynth/cforth. This is a fork of https://github.com/MitchBradley/cforth. The upstream project is a portable Forth for micro-controllers that uses PlatformIO to achieve that portability. My fork is to provide infrastructure for extending this portability the RP2350 boards.

So my next steps are to clean up the README documentation, noting that the fork of `platform-raspberrypi` is required, and then submitting a pull request to upstream. I have no other plans to use PlatformIO.