platformio/platform-raspberrypi #68: Please add support for Seeed Studio XIAO RP2040

Steven-Burns, 2024-08-27:
What kind of issue is this?

- [ X] **Feature Request**.

### Additional info

https://wiki.seeedstudio.com/XIAO-RP2040/

Steven-Burns, 2024-08-27:
Meaning, complete the PRs and integrations necessary to integrate the PR #36 cited in this forum thread 
https://community.platformio.org/t/support-for-seeed-xiao-rp2040/28989/5

febles, 2026-02-16:
Hi there!

+1 on this. Just ran into the same problem.

I'm working through Fab Academy 2026 assignments and wanted to test the Seeed Studio XIAO RP2040 in PlatformIO. The board simply doesn't show up in the Project Wizard or `pio boards` output.

**Current workarounds I've found:**

1. Using maxgerhardt's fork:
```ini
   [env:seeed_xiao_rp2040]
   platform = https://github.com/maxgerhardt/platform-raspberrypi.git
   board = seeed_xiao_rp2040
   framework = arduino
   board_build.core = earlephilhower
```

2. Using Seeed's own platform:
```ini
   [env:seeed_xiao_rp2040]
   platform = https://github.com/Seeed-Studio/platform-seeedboards.git
   board = seeed-xiao-rp2040
   framework = arduino
```

Both work, but they require pointing to external GitHub repos, which is far from an ideal first-time experience — especially for students and newcomers who expect to just pick a board from the official list.

**The XIAO RP2040 is widely used in educational contexts** like Fab Academy, where it's one of the recommended boards for embedded programming assignments. Having it available out of the box in PlatformIO would remove a significant friction point.

The board definition already exists in maxgerhardt's fork (which has been actively maintaining RP2040 board support for years now). Would it be possible to merge these board definitions into the official platform?

This issue has been open since Aug 2024 (and earlier requests go back to 2022 — see #25 and #35). The community interest is clearly there. Would love to see this land in an official release. 

Thank you!!