platformio/platform-raspberrypi #79: Use latest Pico-SDK in raspberrypi platform

jhmaloney, 2025-05-26:
After a clean install of the raspberrypi platform, the "pico-sdk" subrepository is in a detached head state at commit b1676c1.

Is it possible for the raspberrypi  platform to use a later release of pico-sdk? There was a recent change in the SDK support for the RP2350-B board a few months ago. They removed the PICO_RP2350B #define and now use PICO_RP2350A=0 to indicate RP2350.

Those #defines are used to enable the extra GPIO pins on the PICO_RP2350B. I am porting MicroBlocks to the Sparkfun XRP controller board (an educational robotics board).

As a workaround, would it be safe for me to fast-forward the pico-sdk to the current head? Or might that break something?

Thank you!

valeros, 2025-05-27:
Hi @jhmaloney, you're probably reporting a wrong repository as this dev-platform doesn't support Pico SDK.

closed 2025-05-27 by valeros

jhmaloney, 2025-05-28:
Yes, I'm using the alternative Arduino core maintained by  @maxgerhardt. I will ask about this on his repository.

Thanks for pointing me in the right direction.